wled/WLED #557: Constant Reboots

seantibb, 2020-01-05:
I cloned the latest repo in the hopes of deploying it to my own board with some new features for PIR sensors on the same NodeMCU. I setup the network info and mqtt info in the code, uncommented the NodeMCU board in environment, built successfully and then deployed to the NodeMCU. Everything appears to be working, but it is in a constant reboot loop. Now the reboots happen so frequently, I can't get it into a flash mode successfully to flash the bin file and recover the NodeMCU board. Any suggestions?

Happy to submit code, but not sure how it will change things....

brentbrooks70, 2020-01-05:
@seantibb Does your board have 2 buttons (i.e. Reset & Program)? If so, just hold the program button press and release the reset button, your board should be in programmable flash mode. If your board only has a reset button connect GPIO0 to ground and then press and release the reset button or apply power and you should be able to flash it now. Of course, all of this assumes you can plug into the device and are not limited to OTA methods of flashing

Good luck

Aircoookie, 2020-01-05:
Hi!
First of all sorry for the bootloop issue and I hope you'll be able to reflash your ESP to make it work again.
I believe the reboots are caused by an MQTT issue I have yet to work out. What you can try though is setting the network and MQTT settings via the web interface (connecting to WLED-AP) instead of in-code. It miight be that something is not initialized correctly.
Thanks!

seantibb, 2020-01-05:
Thanks for the responses! I've stopped trying to pre-configure and am having no more boot loops. I'll keep my eyes out for an update to resolve this issue and give it a try then. Thanks for all the hard work! Also, if you point me in the direction of the bug, I might be able to help. 30+ years of dev experience ;).

fishbone-git, 2020-01-07:
Could it be the blynk section of code?  I normally have Alexa, HUE, Blynk, etc disabled in the wled00.ino. I recently decided I didn't want to bother with that customization and I have the reboots. As soon as I uncomment only the #define WLED_DISABLE_BLYNK, no more reboots.  Serial log suggests a memory leak (heap too low).  I don't even have blynk enabled in the config of the device, just the code.

seantibb, 2020-01-07:
Good catch on the Blynk piece! I think I commented that out, but not sure. I will flash again tonight or tomorrow with that specific line commented and then not and see if I get the same.

Def3nder, 2020-01-17:
@seantibb did commenting out blank made a change ?

triggerx, 2020-04-22:
Also having constant boot loop issues.... i don't use MQTT.  0.8.6 works completely fine for me, but have never been successful with 0.9.1.  With the 0.9.1 official release .bin, the reboot would happen about every 15 seconds, and be "offline" for about another 15 seconds.  With a compile of current code, the reboot happens about every minute, and is only "offline" for a couple seconds... so it's gotten "better" since 0.9.1.... just not fixed.  Just piling on here in order to track this bug.  i have an exception log if that helps anything.  thanks.

seantibb, 2020-08-25:
FYI....looks like removing Blynk does resolve the constant reboots issue. Not sure what root cause is there....

jbeck22, 2020-11-11:
I'm having the rebooting problem on one of my boards.  How do I go about removing the Blynk configuration?  I don't use Blynk so I'm happy to remove it, just don't know where to do it.

closed 2021-06-06 by stale[bot]