wled/WLED #559: usermod html integration

huggy-d1, 2020-01-06:
Where can we add usermod specific html for proper integration and rendering?

Aircoookie, 2020-01-06:
Hi!
That is actually a tricky topic. You do have the `/u` subpage where it is possible to add a separate HTML page easily (just add it to `PAGE_usermod` in the top of html_other.h).
As for adding content for to the actual UI, there is no convenient method for that currently unfortunately.
You'd need to edit the UI HTML, which also means that the usermod UI version won't have the newest features if I add some unless you kept it up to date manually, which is not acceptable (at least it wouldn't be for me).
I really need to come up with a way to add own content (at least user-defined info displays and buttons) to the UI in a more straightforward or modular way...

sansillusion, 2020-01-06:
Mabe a javascript that would insert a user html stored in spiff. It could
fetch a setting status to see if it should load it.

sansillusion, 2020-01-06:
It could load it from the already existing option for mini web file server
(fs editor). So we could upload it to the esp at will.

huggy-d1, 2020-01-07:
If I understand you, I don't understand you / the documentation / file layout.
If I edit html_other.h such that it refers to htm files named:
other.htm
other_xxx.htm
other_yyy.htm
other_zzz.htm
where the other.h and other.htm offer (in this case) 3 *unique* configuration pages.

Will WLED recognize them _other_ htm files as available and render automatically?

huggy-d1, 2020-01-07:
Here's what I currently have in html_other.h for the user html:
`//USER HTML HERE (/u subpage)
const char PAGE_usermod[] PROGMEM = R"=====(<!DOCTYPE html>
<html><head><meta content='width=device-width' name='viewport'>
<title>WLED QuinLED Dig-Uno Settings</title>
<style>body{text-align:center;background:#222;height:100%;margin:0}html{--h:11.55vh}button{background:#333;color:#fff;font-family:Verdana,Helvetica,sans-serif;border:.3ch solid #333;display:inline-block;font-size:8vmin;height:var(--h);width:95%;margin-top:2.4vh}</style>
<script>function BB(){if(window.frameElement){document.getElementById("b").style.display="none";document.documentElement.style.setProperty("--h","13.86vh")}};</script></head>
<body onload=BB()>
<form action=/><button type=submit id=b>Back</button></form>
<form action=/other/diguno_temp><button type=submit>Temperature Setup</button></form>
<form action=/other/diguno_motion><button type=submit>Motion Setup</button></form>
<form action=/other/diguno_application><button type=submit>Motion Application</button></form>
</body></html>)=====";`
I created /data/other_diguno_application.htm, /data/other_diguno_motion.htm, /data/other_diguno_temp.htm to work with the above as the targets for the setup buttons.
I don't have code in the other_diguno_*.htm done because I'm not sure if this is going to work, and whether it will be fluid and properly integrated into the existing UI.

On top of it all, I cannot easily test it because I have no idea how to get it to compile.

FYI, I'm using VS Code with several plug-ins, but do not have the arduino software installed (on my work PC, IT Policy prevents arduino software from installing and/or launching).

Once I know I'm on the correct path, I can generate code during lunch on the work PC, push to a personal github, and compile it on the home PC.

closed 2020-05-13 by stale[bot]