wled/WLED #561: BGR color order option

mthorwart, 2020-01-07:
I flashed my D1 with 0.9.0 ota, and hooked up some new lights today and couldn’t get them to the correct colors using the color orders. In xlights I was able to determine that they are bgr, but would rather control it in wled. Is there an easy fix? I’m still new to pixels, and esp8266, but I’ll probably dive a little deeper into the coding this weekend anyway.

tube0013, 2020-01-07:
Under the LED preferences in the config section of the wled web interface you can change the color order to align with the lights you have.

Aircoookie, 2020-01-07:
Hi!
Interesting how the manufacturers could agree on a protocol but not on the d**n order the colors are in! Will add BGR order to the color order list :)

mthorwart, 2020-01-07:
> Hi!
> Interesting how the manufacturers could agree on a protocol but not on the d**n order the colors are in! Will add BGR order to the color order list :)

Yeah I was pretty surprised that that is a thing myself. Haha
Just make rgb lights rgb and we don’t have this problem! 
Thanks for the quick reply!

GaryOkie, 2020-01-07:
Thanks a bunch for adding ALL possible combinations of RGB color sequences as an enhancement. 

As I mentioned in #529, my Alitove WS2811 12V strip requires yet a different **GBR** sequence!

Sure looking forward to the update.

Aircoookie, 2020-01-14:
This is now part of latest master!

closed 2020-01-14 by Aircoookie

mthorwart, 2020-01-14:
So dumb question sorry, do I just reflash with the same release? Thank you very much!

Aircoookie, 2020-01-14:
@mthorwart no, that won't work. The release is set and will never change.
You would need to download the latest source code and compile it yourself.
Therefore, here is a binary for the latest version which supports BGR: https://drive.google.com/open?id=1jEKJc6F8KCowfzbOMoItovhqqK7KZdW-

mthorwart, 2020-01-14:
Thank you! Obviously I am still learning all of this, but I am enjoying the hell out of it. 
Thanks again for getting this fixed so quickly

GaryOkie, 2020-01-15:
Just now updated the firmware with your  **0.9.0-b2** binary - works great!   Thanks so much for adding this expanded color sequence option.

BTW, it turns out my strips also needed the BRG sequence, same as OP.   The vendor for the strips had confirmed that they require a GBR sequence but apparently that is only true when they are connected to a different controller that uses MagicHome app (which I have gotten rid of and replaced them with ESP8266's & WLED!)

mthorwart, 2020-01-15:
I also flashed a new chip this morning and everything looked good so far. Working on some soldering then I will hook up the lights for the real test. Thank you!

raoulteeuwen, 2020-01-15:
Flashed the new file to a nodemcu... while switching between effects at some point (while selecting the fireworks starburst effect) got a "TypeError: s.nl is undefined". I ignored it and could continue.

Aircoookie, 2020-01-15:
@raoulteeuwen oh, that sounds like we are potentially running out of memory (or have heap fragmentation)... How many LEDs were you using? (Fireworks starburst, as well as a number of other effects now dynamically allocate memory for saving stuff like particle positions and speeds)

raoulteeuwen, 2020-01-16:
@Aircoookie it's a short strip with 120 LEDs. Maybe me switching effects rapidly also played a role? In the 1st 0.9.0-version while switching effects rapidly, you also got (another, i believe Type, reported in some issue(s))Error...