wled/WLED #562: null pixel support

fhscholl, 2020-01-07:
I'm looking for something like "Skip First LED" but with multiple, user provided pixel indexes.

Here is my example/use case:
I built my own custom landscape lighting by replacing the guts of off-the-shelf lights with groups of 10 12V WS2811 bulbs in a custom mount. I basically have 17 groups of 10 lights wired in series with underground wire. I plan to use the grouping ability in the pending PR to treat these as 17 single "virtual pixels".

I ran into an issue with signal transmission because some of the spacing between lights are too long (turning corners, different beds separated by sidewalk). I'm going to correct this by burying a null pixel in a waterproof box to retransmit the signal. I don't really care if the hidden pixel lights up, but adding the pixel will throw off the count of the grouping.

What I'm looking for is a way to provide pixel numbers to ignore in grouping and sequencing so the visible pattern is correct. (e.g. 0,111,142  where absolute pixel 1 would be 0 and 172 would be 169).

I might try to attempt this myself, but I need to blow some heavy dust off some c++ and arduino brain cells.

Aircoookie, 2020-01-07:
Hi!
Sorry if i'm mistaken, but I believe the feature you need is called spacing :)
It will be availabe shortly together with the upcoming grouping support and allows you to turn off x LEDs between each group. This, for example would be grouping 3, spacing 1, with one LED being of between groups:
```
000-111-222-333-444
```

fhscholl, 2020-01-07:
I don't have a null pixel between every light though. It's more like 000111222-333444-555

Aircoookie, 2020-01-07:
That makes it a bit more complicated indeed! The easiest solution would be to use segments, but of course that means that you can't run an effect across all the lights.
Since this is very specific, I don't think it would make much sense as a general addition to the library.
For your specific case, it should be easily possible to add though. We basically just need to add something like this in setpixelcolor and getpixelcolor methods:
```
if (i > 110) i++; //skip pixel 111
if (i > 141) i++; //skip pixel 142
```

Let me get the general grouping support merged in, then we can see what whould be the easiest way to get your setup working :)

Budman1758, 2020-01-07:
Perhaps something along these lines would be helpful?
[https://www.amazon.com/SUPERNIGHT-Repeater-Signal-Amplifier-Strip/dp/B00E4JQDKE](url)

Def3nder, 2020-01-08:
@Budman1758 the product you linked is for analog RGB stripes, not for digital (addressable) ones 😄

fhscholl, 2020-01-08:
I wanted to give a quick update on how I decided to fix this for now. 

I took one of the landscape lights next to the problem area and removed a pixel from it, so instead of 10 pixels in that light it has 9. I then, in a little sealable box, wired in an extra pixel about halfway along the run where the signal was dropping. 

This works out for grouping because the missing pixel in the light will be the null pixel and nothing else really needed to change.

I'm considering taking some pictures/video and writing up a summary of this project in the HA forums once the grouping/spacing code is released. Overall it was a very interesting learning experience, and I may do more with led pixel lighting in the future.

closed 2020-01-08 by fhscholl