wled/WLED #563: nodemcu not working in 0.9

yuvalabou, 2020-01-07:
After having problems connecting to WLED from local network (It has connected to the router and pinged) I have flashed my nodemcu with v0.85 and it worked flawlessly.
Tried to do an OTA to v0.9 but it failed again to connect on lan.

Aircoookie, 2020-01-07:
Hi and sorry for the issues.
What router model do you have? Is `WLED_0.9.0-b1_esp8266.bin` the file you used?

Mariu86, 2020-01-07:
I have the same problem with NodeMCU, I have 3 NodeMCU working at this moment and they have problems showing in the Android app and in browser the strange thing is that they work because if I use the ir remote they start and change color and effects. @Aircoookie can you please explain how to use the diagnosis in Wled it's nothing in the wiki about this. Thank you again for this great project!

Aircoookie, 2020-01-07:
@Mariu86 thank you, that is really helpful! Also good that the issue also happens when you compile from source, I was afraid it would only happen with binaries, making it really hard to debug.

First of all, uncomment line 33 (in latest master source) of wled00.ino (`#define WLED_DEBUG`), and after flashing via USB open serial monitor to see the output. If it is interesting, you can send it to me! It will print all sorts of info like whether the connection was successful. You can add own debug printouts using `DEBUG_PRINTLN()` if you'd like. Regular `Serial.println()` works as well though if it is temporary :) Also, please check in the Tools menu which settings you are running. I'd recommend ESP8266 arduino core 2.6.3, Lwip 1.4 Higher Bandwidth and Flash 4M (1M SPIFFS).

I've had an RT-N56U as well a few years ago! (though i believe i flashed it wish ddwrt). Maybe i can find it on the attic and see if I can reproduce the issue using it!

Quato38, 2020-01-07:
I have same issus with my Asus routers, Wled connects, but dissconnects after a while.
Asus RT-AC88U is the router i have.

yuvalabou, 2020-01-07:
> Hi and sorry for the issues.
> What router model do you have? Is `WLED_0.9.0-b1_esp8266.bin` the file you used?

yes this is the file.
I have tiglin ac router that I recieved from the ISP

pepeEL, 2020-01-07:
for me WLED works correct with my nodeMCU but after flash nodeMCU new firmware. After that all works perfect.

Def3nder, 2020-01-08:
When reboots appear it's a good option to start clean and clear the flash before flashing again.

For ESP8266 chips you can use the ESP8266 flasher and toggle the "clear flash" switch.
This is the download URL: http://www.dietrich-kindermann.de/Downloads/ESP8266-Flasher-x32-Installer.zip 

When using PlatformIO you can trigger this with 
`platformio run --target erase`
(remember to use the path to platformio.exe: the whole command should look like this:
`C:\Users\[UserName]\.platformio\penv\Scripts\platformio.exe run --target erase`

yuvalabou, 2020-01-09:
Tried to erase the flash every time but had no luck...

Def3nder, 2020-01-09:
@yuvalabou did you try to position the ESP at another place ?
The ESPs without an external antenna (with PCB antenna only) need a realy good WiFi signal ...

yuvalabou, 2020-01-10:
It was literally next to my router..

pjeterinfo, 2020-01-10:
I have 2 ubiquiti accesspoints configured with a 2.4 and 5Ghz SSID. Both networks share the same SSID and wled is connecting to the accesspoint close to the D1-mini. 
At random times the D1-mini is suddenly offline but the effect which was active is still runnning, and only way to reconnect is to reset the D1-mini.

As a test I reboot the accesspoint and I see the D1-mini is not able to roam to the other accesspoint, even it is within range. All my other network devices are roaming without any issues.

Now I created an additional SSID  2.4Ghz on one accesspoint only, and the D1-mini is online for 4 days without issues. I believe wled has issues reconnecting to the wireless network if it was disconnected.

I will try if I can reproduce this with debug enabled.

closed 2020-05-18 by stale[bot]