wled/WLED #567: Segments lost after reboot

AlphaJosh, 2020-01-07:
I am trying to implement a toilet door occupancy indicator using three WS2812B LEDs attached to a Wemos D1 Mini 

To control each single LED, I have created 3 segments and from the gui I can control each individual LED:
Segments: 
1) LED 0 -1
2) LED 1 -2
3) LED 2 -3

However, when I reboot the D1 mini, my segments are reset. 
How do I configure the segments to survive a reboot please?

Also, does anyone know how I can get Home Assistant to push the lights on each segment to go Green, Yellow or Red?

Def3nder, 2020-01-08:
Hi @AlphaJosh with Home Assistant you can send a JSON object to WLED.
(see https://github.com/Aircoookie/WLED/wiki/JSON-API)

E.g. setting the three segments to different colors would be (the colors are in the col" object):
```
{
  "state": {
    "on": true,
    "bri": 255,
    "seg": [
      {
        "start": 0,
        "stop": 1,
        "len": 1,
        "col": [
          [
            255,
            0,
            0,
            0
          ]
        ],
        "fx": 0
      },
      {
        "start": 1,
        "stop": 2,
        "len": 1,
        "col": [
          [
            255,
            255,
            0,
            0
          ]
        ],
        "fx": 0
      },
      {
        "start": 2,
        "stop": 3,
        "len": 1,
        "col": [
          [
            0,
            255,
            0,
            0
          ]
        ],
        "fx": 0
      }
    ]
  }
}
```

But: why do you use **three** LEDs ? You could use **a single one** displaying either RED, YELLOW or GREEN 😄

CosmicMac, 2020-01-08:
As a colour-blind person, I'm quite satisfied to have traffic lights with 3 positions, in addition to the 3 colours. ;)
Now the real question is: what is the meaning of the color yellow in the context of a toilet door occupancy indicator? :D
BTW, this is probably the tiniest project based on WLED, which illustrates how versatile this fantastic tool is.

Aircoookie, 2020-01-08:
@AlphaJosh you can save one complete segment config in Preset slot 16. Then you can set that as boot preset in LED settings :) I'm working to increase the number of presets that can save segments in 0.10.0!

@CosmicMac it's close to the tiniest! I have an electronic candle running just a single LED with WLED :D

sr1sws, 2020-01-08:
"Now the real question is: what is the meaning of the color yellow in the context of a toilet door occupancy indicator? :D"...

"Exercise caution upon entering confined space."

AlphaJosh, 2020-01-08:
There are three toilets and so I require 3 leds to indicate their use:
Red - Occupied
Green - Free
Yellow - Possible Bio Hazard - Previous occupant has been in for more than 15 mins. This will change to green after 5 minutes.

AlphaJosh, 2020-01-08:
@Aircoookie
> @AlphaJosh you can save one complete segment config in Preset slot 16. Then you can set that as boot preset in LED settings :) I'm working to increase the number of presets that can save segments in 0.10.0!
> 
I have tried to add them to the Presets and added the preset to the start up but found the segments were not present after rebooting. 

@Def3nder - Can I use the JSON and segments to control a single LED independently from the others in Home Assistant? As one door indicator changes, it shouldnt change the others.

Aircoookie, 2020-01-08:
@AlphaJosh did you try Preset number 16 specifically? Only that one is capable of saving segments.

AlphaJosh, 2020-01-08:
Oooo I didnt realise that. Testing now...

AlphaJosh, 2020-01-08:
@Aircoookie Thank you for that. I misread it as saying you can save in the 16 presets and not specifically preset 16.
Thank you for clarifying.

Any chance you might have the code for Home Assistant or JSON that would change segment 1 colour and not affect any of the others please?

AlphaJosh, 2020-01-08:
For public reference:

Note: Using Xiaomi Door Sensors + ZigbeeMQTT (out of scope of this document)

In WLED:
1) set up multiple segments
2) Saved Profile to slot 16
3) In Config set boot to load profile 16
4) In Config set LED number to 3
Note: MQTT not used

In Home Assistant:
1) Remove any previous WLED intergration (so old segments to be refreshed)
2) Add WLED integration
3) In configuration:
`automation:
  - alias: South Door 1 - Open
    trigger:
      - platform: state
        entity_id: binary_sensor.0x00158d0003589e7c_contact
        to: 'on'
        from: 'off'
    action:
      - service: light.turn_on
        data_template:
          entity_id: light.wled_1
          rgb_color:
            - 0
            - 255
            - 0
          brightness: 255        
          
  - alias: South Door 1 - Close
    trigger:
      - platform: state
        entity_id: binary_sensor.0x00158d0003589e7c_contact
        to: 'off'
        from: 'on'
    action:
      - service: light.turn_on
        data_template:
          entity_id: light.wled_1
          rgb_color:
            - 255
            - 0
            - 0
          brightness: 255
      - service: counter.increment
        entity_id: counter.hour_sth_door1
      - service: counter.increment
        entity_id: counter.day_sth_door1          
        `

Def3nder, 2020-01-09:
> @Def3nder - Can I use the JSON and segments to control a single LED independently from the others in Home Assistant? As one door indicator changes, it shouldnt change the others.

yes, you can either use a specific segment in the JSON or send a JSON with all segments.

So I would suggest to have 2 JSONs: 

- one with seg1 in RED / seg2 in black
- one with seg1 in black / seg in green

closed 2020-05-15 by stale[bot]