wled/WLED #572: wled06_usermod.ino,  wiki changes, and global usermod variables accessible via UI

huggy-d1, 2020-01-08:
Currently wled06_usermod.ino contains these functions:
- void userSetup()
- void userConnected()
- void userLoop()
The wiki was recently updated (in advance?) of changes to this file?
Should wled06_usermod.ino now contain these functions?
- userBeginPreConnection()
- userBegin()
- userLoop()

What about code and compiler directives that need to be run to select usermod options, like if ESP8266 is defined, use these PINs and if ESP32 is defined, use a different set of pins?  Or choosing _how_ input devices are configured, like choosing units for a temperature sensor results in different functions & pieces of logic being compiled / executed.

Perhaps there is a section in wled00.ino for usermod globals to execute?

Or by simply putting code and directives _outside_ of the 3 functions (whatever they are named) causes them to be pre-processed (for compiler directives) and executed at runtime to configure _global_ values so the UI can interact with them?

huggy-d1, 2020-01-08:
Is there a file "WLED06_usermod.h" for inserting globals and such, similar to NbpWrapper.h?

Aircoookie, 2020-01-08:
Hi,
it is actually the other way around - I forgot to update the wiki with the new method names!
Done that just now.

You can access any global variable defined in wled00.ino. You can also insert more inside the usermod file. There is no `WLED06_usermod.h` currently, but variables you add to the top of the usermod should also be accessable globally.

To find out the platform, you can use
```cpp
#ifdef ESP8266
 //esp8266 only code
#else 
 //esp32 only code
#endif

#ifdef ARDUINO_ARCH_ESP32
 //also esp32 only code
#endif
```

huggy-d1, 2020-01-08:
Untested but here's what I have _outside_ the 3 stated functions() in wled06_usermod.ino:
`// `
`// This file allows you to add own functionality to WLED more easily`
`//...`
`//`
`
//Use userVar0 and userVar1 (API calls &U0=,&U1=, uint16_t)`

`// comment this out and nothing works in this usermod file (for troubleshooting purposes)
#define QUINLED_DIG_UNO`

`//`
`// QuinLED Dig-Uno boards have extra capabilities available for use with WLED`
`//  - including a Dallas Temperature sensor, DS18B20, which can report board`
`//    temperature via MQTT periodically.  The sensor is connected to IO14 (ESP8266/D5),`
`//    or IO18 (ESP32)`
`//  - including 4 IO additional IO pins as follows:`
`//    Pin   ESP8266         ESP32           Description`
`//    Q1    IO5/D1          IO22            CAN BE PULLED LOW (USING DIG-UNO CKT BD JMPR)`
`//    Q2    IO4/D2          IO21            CAN BE PULLED LOW (USING DIG-UNO CKT BD JMPR)`
`//    Q3    IO12/D6         IO19            CAN BE PULLED HIGH (USING DIG-UNO CKT BD JMPR)`
`//    Q4    IO13/D7         IO13            CAN BE PULLED HIGH (USING DIG-UNO CKT BD JMPR)`
`//  - L1_C  IO1/TX          IO1/TX          ADDRESSABLE LED CONTROL OUTPUT [FOR ESP32 / SECOND STRIP]`
`//  - L1_D  [IO2]|[IO3]/RX  [IO2]|[IO3]/RX  ADDRESSABLE LED CONTROL OUTPUT [JUMPER SELECT GPIO2|GPIO3]`
`//`

`#ifdef QUINLED_DIG_UNO`
`  // detect which processor binary is being compiled for QuinLED Dig-Uno board`
`  #ifdef ESP8266`
`    // ESP8266 binary being compiled for QuinLED Dig-Uno board`
`    // optional Q I/O on Dig-Uno board`
`    #define QUINLED_DIG_UNO_Q1        5     // pin D1, IO Pin 5  <I2C>`
`    #define QUINLED_DIG_UNO_Q2        4     // pin D2, IO Pin 4  <I2C>`
`    #define QUINLED_DIG_UNO_Q3        12    // pin D6, IO Pin 12 <SPI>`
`    #define QUINLED_DIG_UNO_Q4        23    // pin D7, IO Pin 13 <SPI>`

`    // which pin connects to LED for this binary build?`
`    #if LEDPIN      == 2`
`      #define QUINLED_DIG_UNO_LED     2     // pin D4, IO Pin 2`
`    #elif LEDPIN    == 3`
`      #define QUINLED_DIG_UNO_LED     3     // pin RX, IO Pin 3`
`    #endif`
`     // what pin on the ESP8266 is connected to the DS temp sensor?`
    
`    #define QUINLED_DIG_UNO_TEMP      14    // pin D5, IO Pin 14 <SPI>`
`  #endif`

`  #ifdef ARDUINO_ARCH_ESP32`
`    // ESP32 binary being compiled for QuinLED Dig-Uno board`
`    // optional Q I/O on Dig-Uno board`
`    #define QUINLED_DIG_UNO_Q1      22    // pin P22, IO Pin 22 <I2C SCL>`
`    #define QUINLED_DIG_UNO_Q2      21    // pin P21, IO Pin 21 <I2C SDA>`
`    #define QUINLED_DIG_UNO_Q3      19    // pin P19, IO Pin 19 <VSPIMISO>`
`    #define QUINLED_DIG_UNO_Q4      13    // pin P13, IO Pin 13 <ADC14> <TOUCH4>`
    
`    // which pin connects to LED for this binary build?`
`    #if LEDPIN    == 2`
`      #define QUINLED_DIG_UNO_LED   2     // pin P2, IO Pin 2 <ADC12> <TOUCH2>`
`      #define QUINLED_DIG_UNO_LED2  3     // pin RX, IO Pin 3 <RX0>`
`      #define LEDPIN2               3     // second LED strip support`
`    #elif LEDPIN  == 3`
`      #define QUINLED_DIG_UNO_LED   3     // pin RX, IO Pin 3 <RX0>`
`      #define QUINLED_DIG_UNO_LED2  2     // pin RX, IO Pin 3 <RX0>`
`      #define LEDPIN2               2     // second LED strip support`
`    #endif`
    
`    // what pin on the ESP32 is connected to the DS temp sensor?`
`    #define QUINLED_DIG_UNO_TEMP    18    // pin P18, IO Pin 18 <VSPI SCK>`
`  #endif`
`#endif`

`/*`
` *  Comment out QuinLED Dig-Uno UserMod options as required`
` */`

`#ifdef QUINLED_DIG_UNO`
` #define USERMOD_QUINLED_DIG_UNO_TEMP_MQTT`
` #define USERMOD_QUINLED_DIG_UNO_PIR_CONTROL`
`#endif`

`/*============================================`
` * QuinLED DigUno Temp Sensor userSetup`
` */`
`  //Initiating code for QuinLED Dig-Uno temp sensor`
`  #ifdef USERMOD_QUINLED_DIG_UNO_TEMP_MQTT`
`    #include <DallasTemperature.h>`
`    OneWire oneWire(QUINLED_DIG_UNO_TEMP); `
`    DallasTemperature sensors(&oneWire);`
`    long tempTimer = millis();`
`    long lastTempMeasured = 0;`
`    //Uncomment Celsius if that is your preferred temperature scale`
`    //#define Celsius`
`  #endif`
`/*============================================*/`

`/*============================================`
` * QuinLED DigUno PIR Control userSetup`
` * `
` * `
` */`
`  #ifdef USERMOD_QUINLED_DIG_UNO_PIR_CONTROL`
`    #define APPLICATION         2               // 0=No motion sensors, 1=1 motion sensor to activate strip, 2=2 motion sensors for stairway lighting `

`    /* motion detectors */`
`    #define TOP_MOTION_PIN      QUINLED_DIG_UNO_Q1`
`    #define BOTTOM_MOTION_PIN   QUINLED_DIG_UNO_Q2`

`    /* # motion detectors being supported */`
`    #ifdef TOP_MOTION_PIN && BOTTOM_MOTION_PIN`
`      byte MOTION_INPUTS = 2;`
`    #elif TOP_MOTION_PIN || BOTTOM_MOTION_PIN`
`      byte MOTION_INPUTS = 1;`
`    #else`
`      byte MOTION_INPUTS = 0;`
`    #endif`

`    /* stairway lighting direction */`
`    typedef enum {
      DIRECTION_TOP_TO_BOTOM,
      DIRECTION_BOTTOM_TO_TOP,
      DIRECTION_BOTH,
      DIRECTION_NONE
    } direction_t;`

`  #endif`
`/*============================================*/`
`

closed 2020-05-21 by stale[bot]