wled/WLED #576: Home assistant not setting effects to synced devices

geyles, 2020-01-09:
I can go to the devices IP GUI and set the effect which inturn sets the effect of the others but from home assistant if I set the effect it will only change the one device. If I then select a colour (while the effect is on it will then change all of them). If I change solid colour it does work fine though.

Def3nder, 2020-01-09:
Hi @geyles , what did you slect in the Sync settings in section **WLED Broadcast** ?
Send notifications on direct change: ☑️ 
Send notifications on button press: ☑️ 
Send Alexa notifications: ☑️ 
Send Philips Hue change notifications: ☑️ 
Send Macro notifications: ☑️ 
Send notifications twice: ☑️

geyles, 2020-01-09:
I already have all them ticked
[image]

geyles, 2020-01-11:
Also the built in GUI when choosing solid colours it smoothly transitions to the new colour. When using home assistant to change solid colour it instantly changes.

geyles, 2020-02-25:
Now seems to work in the latest release

closed 2020-02-25 by geyles