wled/WLED #578: Effect "Police All" works different with >100 LEDs

Def3nder, 2020-01-09:
With less then 100 Pixels the effect does fill the whole strip solid with RED / BLUE...
...but with more Pixels it's working different:

See this video: https://www.youtube.com/watch?v=wDoUTj43Q94

I tried to change the build flags from PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH (v2) to PIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH (v1.4) but this did not have any effect.

Any suggestion how to change this ?

Aircoookie, 2020-01-09:
Hi!
Thank you for making me aware of this! This problem happens when the effect speed and/or LED count is high enough to move the effect by more than one pixel every frame. That will lead to some of the pixels not getting filled (and then potentially getting filled with the second color, which effectively makes the effect purple). Will fix soon!

Aircoookie, 2020-01-09:
also happens with Sinelon by the way, and Juggle (although the effect is quite nice with juggle ang i wouldn't neccessarily change that)

Dreamoffice, 2020-01-19:
@Aircoookie 
i have another issue with Police All Effect. with 100 LED in the preferences it works like a charm but with more (i have 816) it works like on the video.
[https://youtu.be/ESaA0FjHbnU](url)

Def3nder, 2020-01-28:
closed with PR #634

closed 2020-01-28 by Def3nder