wled/WLED #579: Feature requests: trigger effects [ ], UI set # segments, bidirectional fill / wait / clear effect

huggy-d1, 2020-01-09:
Trigger option to start macro preset (like from motion sensor, door switch, light switch, analog value - like light level, humidity, temperature, wind, pressure, and such) available to usermods

New effect that can perform, once triggered, bidirectional actions similar to as shown in this youtube: https://youtu.be/NHkju5ncC4A

Maybe it's possible to set the number of segments from usermod?  If so, maybe a macro preset using this new effect,
<FX.h>
#define FORWARD                     true
#define REVERSE                        false
#define FX_MODE_FILL_FWD     100
#define FX_MODE_FILL_REV      101
#define FX_MODE_HOLD          102
...
_mode[FX_MODE_FILL_FWD]     = &WS2812FX::mode_fill_fwd
_mode[FX_MODE_FILL_REV]      = &WS2812FX::mode_fill_rev
_mode[FX_MODE_HOLD]          = &WS2812FX::mode_hold

<FX.cpp>

uint16_t WS2112FX::mode_fill(bool direction, RgbwColor fillcolor, uint16_t segment_ID) {
// requested new effect
}
uint16_t WS2112FX::mode_fill_fwd(RgbwColor fillcolor, uint16_t segment_ID) {
  return mode_fill(FORWARD, fillcolor, segment_ID)
}
uint16_t WS2112FX::mode_fill_rev(RgbwColor fillcolor) {
  return mode_fill(REVERSE, fillcolor, segment_ID)
}
uint16_t WS2112FX::mode_hold(uint_16 hold_msec) {
// requested new effect to just leave strip alone for # of seconds or msec or ss.s
}
------------------
Want something like Macro 1 that can be triggered:
fill segment 0 forward with some color
hold for some amount of time
fill num segments -1 reverse with black (no color)

Macro 2 that can be triggered:
fill number of segments -1 reverse with some color
hold for some amount of time
fill segment 0 forward with black (no color)

Anyhow, the video expresses one possible end result.

Is there a way to iterate one segment (step) at a time (one segment per step - max of 39 steps if I read memory limits correctly), kind of like in the video where it pauses around mid-way at a platform, where it there were 2 speeds: segment (stair) speed and overall speed which establishes when each segment activates an effect?

I think it could be quite visually stunning and not too awfully difficult now that segments are part of WLED.
Thoughts welcome by any and all!  I am thinking of lots of places and uses for this other than just stairs.

Aircoookie, 2020-01-10:
Hi!
Yes, I do believe that those staircase lights would be amazing running WLED.
Probably this would be best as a usermod for now. We don't even need to add any new effects, just have Wipe run once cycle and switch to Solid automatically. Wait for x seconds, then reverse the direction and switch to wipe to turn it off again.
Obviously multiple segments like in the video are possible as well, but that needs to be coded individually.

I will share a few code snippets tomorrow that show how you  can easily change effects from the usermod code :)

Here is another cool video with LEDs under each stair with each one turning on in sequence. That will be possible with segment grouping, which will be part of 0.9.1 :)
https://www.youtube.com/watch?v=dROAjykbJyQ

miraixyz, 2020-01-10:
Would definitely love to see this too!!

huggy-d1, 2020-01-10:
Ok, wled06_usermod_LinearEffectLighting aka Stairway, under cabinet, under toe-kick of floor-cabinets, under bed, around closet doors, along hall-way...

// pseudo code to fill segments (steps for stairway)
for id = start_segment_id to stop_segment_id 
  wipe( forward, id, color, speed, step_filled(id) )
  // where wipe returns a done bit as true upon filling segment with color
  if step_filled(id)
    delay(xxx)
    or use if "enough_time_has_elapsed" then exit loop

// now we wait for person to complete whatever task is being lit,
// which waits for a trigger to signal time to clean up
// could be wait until external input + some time, wait for some trigger, just wait for x amount of time

if trigger_to_cleanup then 
  for id = stop_segment_id to start_segment_id
    wipe( reverse, id, black, speed, step_filled(id) )
    if step_filled(id)
      delay(xxx)
      or if enough_time_has_elapsed then exit loop

I know the above will not work in arduino/C/C++ as written, but it hopefully gets the idea across why I submitted the feature request(s) because how do you delay and how do you detect an effect has completed and how do you start an effect from the beginning or the end whenever you call it?

Aircoookie, 2020-01-10:
I'm currently in the progress of writing a small usermod for this :)
For now it will only support API calls/macros, but it should be relatively easy to add PIR sensor support!

Aircoookie, 2020-01-10:
Here is the usermod! Fully functional, but it doesn't yet have an interface to any sort of sensor.
Turning off via a Wipe instead of fading is also something i'd like to add soon!

https://github.com/Aircoookie/WLED/tree/master/usermods/stairway_wipe_basic

Aircoookie, 2020-01-10:
If you extend the code to interface with sensors, a PR with your changes as a separate usermod would be very welcome :)

huggy-d1, 2020-01-14:
Working on it.  I will need help with that public relations (PR) thing you mentioned.  I tend to steer clear of the public eye ;-)

I do have questions though.

If I want to have UI for the usermod, just html_other.h set to something other than what it is by default, does that fix it?

Nothing in the wiki about how to add UI to a usermod even though html_other seems to indicate it was thought about.

I'm adjusting the usermod so it can support 0, 1, or 2 motion sensors and then activate somewhat appropriately for each setup.

For no sensors, it would rely upon the API to trigger the on/off and wait time.
For 1 sensor, activate LEDs using some preset, maybe preset 1 (or 11), then stay on for the preset wait time (or preset via API)
For 2 sensor, activate LEDs with direction depending on which sensor is active, and if both become active, it wipes ON in both directions, and wipes OFF direction & wait time depending on whichever sensor went off last.

Do usermods have flash storage available for storing API configurations like /u0=___ and /u1=____ so after power cycles, settings are recalled?

Def3nder, 2020-01-17:
In the v2 of the stairway time-of-flight sensors are used:

VL53L0X Time-of-Flight (ToF) Laser Ranging Sensor Breakout 940nm GY-VL53L0XV2 Laser Distance Module I2C IIC 3.3V/5V For Arduino

...can get bought here: 
https://de.aliexpress.com/item/32960610502.html?spm=a2g0s.9042311.0.0.17574c4dKwXihT

I will write some code once I've received them (have bought some of them) :smile:

Mariu86, 2020-01-18:
I am interested in this :D thanks @Def3nder !

Aircoookie, 2020-01-18:
Just added the possibility for a reverse wipe effect in https://github.com/Aircoookie/WLED/commit/569efd3e45f354e67bbca88b747812aa6b8bdcd8 !

huggy-d1, 2020-01-19:
Does the reverse wipe darken up or down the strip?  Like on stairs, walking up lightens up, pauses, then darkens from bottom to top. And opposite for walking down the stairs.

Aircoookie, 2020-01-19:
Well yes, the total light output will dim down because there will be less lit LEDs. But the light output for any individual LED will not dim, it's just the second, "backwards" part of the Wipe effect running

Svrbinek, 2020-04-22:
> In the v2 of the stairway time-of-flight sensors are used:
>
> VL53L0X Time-of-Flight (ToF) Laser Ranging Sensor Breakout 940nm GY-VL53L0XV2 Laser Distance Module I2C IIC 3.3V/5V For Arduino
>
> ...can [...]

Hello. I managed to program in Node RED and with the help of WLED and stairway_wipe_basic exactly the effect which is in the video for V2 stairs. it is not necessary to modify the stairway_wipe_basic code either, but it is necessary to create a composite code in the API for win & U0 = 0. Instead, two new ones are used, U0 = 0 & RV = 0 and U0 = 0 & RV = 1. Then it works exactly as it should :-) Enjoy!

bodylinksystems, 2020-05-14:
@Svrbinek Could you please share how you did this in node-red? I can control the lights via node-red but can't get my motion sensor set up.  Did you just add  `pinMode(userVar0, INPUT);` ?

Svrbinek, 2020-05-14:
I don't have sensors in Node Red yet. There are improvised Timestamps, instead of which pulses from sensors will be brought. If you want, I'll send a screen as I have it. I'm still waiting for the delivery of sensors, so I can't implement it yet.

TylersSavage, 2020-06-08:
Has anyone ever gotten two PIR sensors to work? With the wipe going in the direction of travel based on which sensor activates first? I’m using the stairway mod, but I have a single smart things sensor to turn it on and off. I’ve been trying for two inexpensive PIR’s but I’m lost!

TylersSavage, 2020-06-08:
My attempt at the stairway mod, using one smartthings motion sensor and webcore for now. https://streamable.com/sjzayf
Still trying to figure out how to integrate two sensors for on/off, and different start points.

Svrbinek, 2020-10-23:
I keep going and tuning the lights. I encountered some obstacles that I will try to describe. Here are links to two versions for now. So far, none of them are working at 100% due to the fact that I would like to use some spot edge light or a subtle effect in standby mode, when no one is on the stairs. These combinations then occasionally confuse the WLED and then make it malfunction.
version 1:
https://www.youtube.com/watch?v=UH_S_P3kfPA
version 2:
https://www.youtube.com/watch?v=trqgV3U3gKA

The problem is that you can't combine brightness commands with stair commands.
For example, this one is not working, it happens that the brightness changes in the setting, WLED looks like it is turned on according to the switch, but the effect does not work. Then the whole WLED is a bit bitten and you can't run another effect.
example: U0=1&A=1

blazoncek, 2022-10-10:
It is possible to modify every aspect of WLED from a usermod. Including getting information from outside sources such as MQTT or HTTP requests.
Samples available in included usermods (Animated Staircase, Multi relay, PIR sensor, etc).
Available since 0.12 and enhanced in 0.13 and forthcoming 0.14.

closed 2022-10-10 by blazoncek