wled/WLED #581: "Disable repeating N LEDs" feature request, make this an effect option

snook, 2020-01-10:
Thank you for adding this feature. Currently on the effect options there is Effect speed, intensity and mode, it would be nice to have this as a 4th option per effect and able to be saved per favorite rather than as a global setting.

I would like to do accent lighting by using the feature throughout the year, but there are occasions, up to once a week and all month long during the holidays, where I want to use the whole string.

Thanks

Aircoookie, 2020-01-10:
Hi!
This is already in the works and should be added this weekend! It is called Grouping and Spacing.
You can set it per segment, so it works in every effect!
Spacing is just "Disable repeating N LEDs" while grouping allows several adjacent physical LEDs to act as one large virtual LED. That will be useful for high density strips and other applications like Cololight/Nanoleaf clones.

You won't have to wait for long :)

palengh, 2020-02-16:
My string has every second LED dark when “Disable repeating N LEDs:” is set to “0”.
When set to “1” they’re all dark. Switching to “2” dims out 2 consecutive LEDs

snook, 2020-02-16:
> My string has every second LED dark when “Disable repeating N LEDs:” is set to “0”.

Should open an issue ticket or find a related ticket, this ticket is regarding a feature request.

Aircoookie, 2020-04-17:
Closing as this was added as segment spacing in 0.9.1 :)

closed 2020-04-17 by Aircoookie