wled/WLED #586: Feature proposal: Temporary Effects/Colors with duration

recliq, 2020-01-13:
Hi, first of all thanks for this great project!

In the software I used before I hacked a little feature which allows to display a color/effect/preset temporarily by passing an additional duration argument. This way the color/effect was displayed the specified amount of time and after that the previous state was restored. 
This would be a nice feature to have on WLED as well...
I use this for notifications to eg. blink a strip for 10 seconds.

Greets 
recliq

Aircoookie, 2020-01-13:
Hi, thank you!

This is a very good idea! I will think about a good way to add something like this :)

Def3nder, 2020-01-14:
Hi @Aircoookie , could this be implemented with an overlay like the clock?

sophof, 2021-07-18:
I've been trying to implement this in Home Assistant with a script. It works fine except if the light is off. Since in home assistant it (for some reason) intended behaviour to not report the effect when a light is off, it is impossible to set it back to the original.

Long story short, I'd really love to have this feature and it is impossible to create an alternative in HA :(

blazoncek, 2023-01-25:
Use playlist feature, available since 0.12 and enhanced in 0.13.

closed 2023-01-25 by blazoncek