wled/WLED #588: Arduino IDE compile errors (ESP-WROOM-32)

andzno1, 2020-01-13:
I am trying to compile the code using the Arduino IDE (1.8.10) and installed all dependencies as decribed in the [Quick start guide](https://github.com/Aircoookie/WLED/wiki#quick-start-guide).

I am using an ESP-WROOM-32 board and installed it in the Arduino IDE using the following url:

`https://dl.espressif.com/dl/package_esp32_index.json`

For compiling and flashing I selected the `FireBeetle-ESP32` in the Arduino IDE.

When compiling I get the following output:

```E:\xxx\yyy\WLED\wled00\wled18_server.ino: In lambda function:

wled18_server:148:66: error: invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

       request->send_P(200, "text/html", PAGE_update, msgProcessor);

                                                                  ^

In file included from E:\xxx\yyy\WLED\wled00\wled00.ino:50:0:

E:\Dokumente\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:234:10: note:   initializing argument 3 of 'void AsyncWebServerRequest::send_P(int, const String&, const uint8_t*, size_t)'

     void send_P(int code, const String& contentType, const uint8_t * content, size_t len);

          ^

wled18_server:148:66: error: invalid conversion from 'String (*)(const String&)' to 'size_t {aka unsigned int}' [-fpermissive]

       request->send_P(200, "text/html", PAGE_update, msgProcessor);

                                                                  ^

In file included from E:\xxx\yyy\WLED\wled00\wled00.ino:50:0:

E:\Dokumente\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:234:10: note:   initializing argument 4 of 'void AsyncWebServerRequest::send_P(int, const String&, const uint8_t*, size_t)'

     void send_P(int code, const String& contentType, const uint8_t * content, size_t len);

          ^

E:\xxx\yyy\WLED\wled00\wled18_server.ino: In function 'void serveMessage(AsyncWebServerRequest*, uint16_t, String, String, byte)':

wled18_server:325:60: error: invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

   request->send_P(code, "text/html", PAGE_msg, msgProcessor);

                                                            ^

In file included from E:\xxx\yyy\WLED\wled00\wled00.ino:50:0:

E:\Dokumente\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:234:10: note:   initializing argument 3 of 'void AsyncWebServerRequest::send_P(int, const String&, const uint8_t*, size_t)'

     void send_P(int code, const String& contentType, const uint8_t * content, size_t len);

          ^

wled18_server:325:60: error: invalid conversion from 'String (*)(const String&)' to 'size_t {aka unsigned int}' [-fpermissive]

   request->send_P(code, "text/html", PAGE_msg, msgProcessor);

                                                            ^

In file included from E:\xxx\yyy\WLED\wled00\wled00.ino:50:0:

E:\Dokumente\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:234:10: note:   initializing argument 4 of 'void AsyncWebServerRequest::send_P(int, const String&, const uint8_t*, size_t)'

     void send_P(int code, const String& contentType, const uint8_t * content, size_t len);

          ^

[... 205 lines cut ...]
 Used: C:\Users\zzz\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\DNSServer
Multiple libraries were found for "ArduinoOTA.h"
 Used: C:\Users\zzz\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA
Multiple libraries were found for "SPI.h"
 Used: C:\Users\zzz\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI
Multiple libraries were found for "FastLED.h"
 Used: E:\Dokumente\Arduino\libraries\FastLED
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\zzz\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program
Multiple libraries were found for "AsyncTCP.h"
 Used: E:\Dokumente\Arduino\libraries\AsyncTCP
Multiple libraries were found for "FS.h"
 Used: C:\Users\zzz\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS
Multiple libraries were found for "ESPAsyncWebServer.h"
 Used: E:\Dokumente\Arduino\libraries\ESPAsyncWebServer
Multiple libraries were found for "Update.h"
 Used: C:\Users\zzz\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Update
Multiple libraries were found for "NeoPixelBrightnessBus.h"
 Used: E:\Dokumente\Arduino\libraries\NeoPixelBus_by_Makuna
exit status 1
invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]
```

Has anybody else experienced this issue? I couldn't find anything. Any help is very much appreciated.

Def3nder, 2020-01-14:
Hi @andzno1 
I would suggest to install VS Code and then the extension "PlatformIO IDE".
If you did this, you only need to open the directory where you cloned the repository in , open the platformio.ini file and uncomment the line "; env_default = esp32dev" in section [platformio] and start the compile (icon ✔️ in the bottom toolbar).

PlatformIO does care about everything else (the correct libraries, the framework, the compiler flag, etc.).

andzno1, 2020-01-14:
Thank you very much for your quick reply, @Def3nder.
This is kind of working, but now I get the following error:

```
Linking .pio\build\esp32dev\firmware.elf
.pio\build\esp32dev\libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x4): undefined reference to `setup()'
.pio\build\esp32dev\libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `loop()'
.pio\build\esp32dev\libFrameworkArduino.a(main.cpp.o): In function `loopTask(void*)':
main.cpp:(.text._Z8loopTaskPv+0x3): undefined reference to `setup()'
main.cpp:(.text._Z8loopTaskPv+0x15): undefined reference to `loop()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

closed 2020-01-14 by andzno1

fishbone-git, 2020-01-14:
it may still fail occationally.  In that case, turn off PlatformIO > Rebuild C/C++ Project Index (Autocomplete, Linter) as it sometimes rebuilds when you are trying to compile.

pruwait, 2020-01-30:
Source code consist some errors.
Its warnings only for PlatformIO, but errors for ArduinoIDE.

Need fix warnings and errors.

This code (version 0.9.0-b2) compiled compiled successfully on Arduino IDE 1.8.10.
But I not using ALEXA, BLYNK, CRONIXIE, MQTT.
[wled00.zip](https://github.com/Aircoookie/WLED/files/4133343/wled00.zip)

Def3nder, 2020-01-30:
hi @pruwait, this issue is closed - so "normaly" nobody would see your comment 😄 
It would help to open new issues instead.

Comming to your problem:

You did not write anything about the errors you get.
Then you write  "but errors for ArduinoIDE" and below "compiled successfully on Arduino IDE 1.8.10".
Which is correct now ?

In general: getting all the dependencies correct is far more complicated in Arduinmo IDE.
PlatformIO does all this for you without any interaction.

What are the errors about in detail?

Def3nder, 2020-01-30:
@andzno1 this happens if you change something in the `platformio.ini` and press compile (✔️) too fast.
PlatformIO does an `InteliSense Index Rebuild` every time the ini changes - this is displayed in the bottom status bar.

Wait this Index Rebuild to be completed and hit "Build" after that.
Then everything works fine.

pruwait, 2020-01-30:
> Then you write "but errors for ArduinoIDE" and below "compiled successfully on Arduino IDE 1.8.10".
> Which is correct now ?

Both correct. I fix errors for myself and add file wled00.zip to post issuecomment-580174709.
You can see all fixes If compare my files with github source.

`Arduino: 1.8.10 (Windows 10), Плата:"WEMOS LOLIN32, 80MHz, Default, 240MHz (WiFi/BT), 921600"

FX.cpp:1385:12: error: narrowing conversion of '(int)(((WS2812FX*)this)->WS2812FX::_segments[((int)((WS2812FX*)this)->WS2812FX::_segment_index)].WS2812FX::Segment::length() / 4u)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]

     {SEGLEN/4,   SEGLEN/8,  1, 1},

            ^

FX.cpp:1385:24: error: narrowing conversion of '(int)(((WS2812FX*)this)->WS2812FX::_segments[((int)((WS2812FX*)this)->WS2812FX::_segment_index)].WS2812FX::Segment::length() / 8u)' from 'int' to 'int8_t {aka signed char}' inside { } [-Werror=narrowing]

     {SEGLEN/4,   SEGLEN/8,  1, 1},

                        ^

FX.cpp:1386:14: error: narrowing conversion of '(((int)(((WS2812FX*)this)->WS2812FX::_segments[((int)((WS2812FX*)this)->WS2812FX::_segment_index)].WS2812FX::Segment::length() / 4u)) * 2)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]

     {SEGLEN/4*2, SEGLEN/8, -1, 1}

              ^

FX.cpp:1386:24: error: narrowing conversion of '(int)(((WS2812FX*)this)->WS2812FX::_segments[((int)((WS2812FX*)this)->WS2812FX::_segment_index)].WS2812FX::Segment::length() / 8u)' from 'int' to 'int8_t {aka signed char}' inside { } [-Werror=narrowing]

     {SEGLEN/4*2, SEGLEN/8, -1, 1}

                        ^

sketch\FX.cpp: In member function 'uint16_t WS2812FX::mode_noise16_2()':

sketch\FX.cpp:1718:14: warning: unused variable 'shift_y' [-Wunused-variable]

     uint16_t shift_y = SEGENV.step/42;

              ^

sketch\FX_fcn.cpp: In member function 'bool WS2812FX::setEffectConfig(uint8_t, uint8_t, uint8_t, uint8_t)':

sketch\FX_fcn.cpp:307:11: warning: unused variable 'mainSeg' [-Wunused-variable]

   uint8_t mainSeg = getMainSegmentId();

           ^

sketch\FX_fcn.cpp: In member function 'void WS2812FX::setIndividual(uint16_t, uint32_t)':

sketch\FX_fcn.cpp:493:9: warning: comparison is always true due to limited range of data type [-Wtype-limits]

   if (i >= 0 && i < _length)

         ^

In file included from sketch\src/dependencies/espalexa/Espalexa.h:60:0,

                 from D:\GoogleDisk\Shared\Arduino\WLED-0.9.0-b1\wled00\wled00.ino:65:

sketch\src/dependencies/espalexa/EspalexaDevice.h:6:15: warning: 'typedef' was ignored in this declaration

 typedef class EspalexaDevice;

               ^

In file included from D:\GoogleDisk\Shared\Arduino\WLED-0.9.0-b1\wled00\wled00.ino:65:0:

sketch\src/dependencies/espalexa/Espalexa.h: In member function 'String Espalexa::typeString(EspalexaDeviceType)':

[... 94 lines cut ...]

                                        ^

D:\GoogleDisk\Shared\Arduino\WLED-0.9.0-b1\wled00\wled05_init.ino: In function 'void wledInit()':

D:\GoogleDisk\Shared\Arduino\WLED-0.9.0-b1\wled00\wled05_init.ino:29:7: warning: unused variable 'heapPreAlloc' [-Wunused-variable]

   int heapPreAlloc = ESP.getFreeHeap();

       ^

D:\GoogleDisk\Shared\Arduino\WLED-0.9.0-b1\wled00\wled17_mqtt.ino: In function 'void publishMqtt()':

wled17_mqtt:82:24: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'int' [-Werror=format=]

   sprintf(s, "%ld", bri);

                        ^

wled17_mqtt:82:24: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'int' [-Werror=format=]

cc1plus.exe: some warnings being treated as errors

narrowing conversion of '(int)(((WS2812FX*)this)->WS2812FX::_segments[((int)((WS2812FX*)this)->WS2812FX::_segment_index)].WS2812FX::Segment::length() / 4u)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
`