wled/WLED #589: ESP8285 crashing after connecting to home ap

wolfman2g1, 2020-01-13:
I've flash with 
WLED_0.9.0-b1_ESP8266_1M_full.bin
WLED_0.8.6_ESP01_1m_full.bin

Everything works until it connect to my wifi I get this message
`[17:48:17]sd[17:53:43]rd[17:54:40]
[17:54:40]Exception (28):
[17:54:40]epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[17:54:40]
[17:54:40]>>>stack>>>
[17:54:40]
[17:54:40]ctx: cont
[17:54:40]sp: 3ffffd30 end: 3fffffc0 offset: 01a0
[17:54:40]3ffffed0:  00000000 00000001 3fff1cc0 3fff2120  
[17:54:40]3ffffee0:  3fffff40 3fffff4c 00000000 4021f26f  
[17:54:40]3ffffef0:  00000050 00000005 3fffff20 3fff1ef8  
[17:54:40]3fffff00:  00000003 3fff2100 3fff1684 3fffff4c  
[17:54:40]3fffff10:  3fffff40 00000050 3fff2120 4021f2da  
[17:54:40]3fffff20:  3ffeaa8b 3fffff40 3ffeaa87 3fff1ef8  
[17:54:40]3fffff30:  3ffeaa8b 3ffeaa87 3fff2120 40215fae  
[17:54:40]3fffff40:  00706374 6704000a 83000000 64656c77  
[17:54:40]3fffff50:  3fff5200 8421da48 401005f8 3fff1ef8  
[17:54:40]3fffff60:  3fff1efd 3fff1ef7 00000000 40216216  
[17:54:40]3fffff70:  00000000 4021da48 3fff22f8 3fff2338  
[17:54:40]3fffff80:  00000000 00000000 00000001 3fff2338  
[17:54:40]3fffff90:  3fffdad0 00000000 3fff22f8 4021be71  
[17:54:40]3fffffa0:  feefeffe feefeffe feefeffe 4022bb8c  
[17:54:40]3fffffb0:  feefeffe feefeffe 3ffe8818 40100465  
[17:54:40]<<<stack<<<
`

raoulteeuwen, 2020-01-14:
Maybe https://github.com/Aircoookie/WLED/issues/323#issuecomment-552227521 works for you?

closed 2020-05-20 by stale[bot]

davidboon78, 2020-08-02:
Hi I'm having this same issue and seem unable to get WLED working on my ESP01

Aircoookie, 2020-08-03:
@davidboon78 Hi, please try deleting the flash (for example using esptool), flash the binary again and then disable the Alexa emulation in Sync settings before entering your WiFi details.
A lot of users have had this issue, so Alexa will be disabled by default in future builds.
Sorry for the trouble!

davidboon78, 2020-08-03:
Hey Sorry aircookie, I wasn't actually getting this issue, it turned out it was the boot looping issue I also commented on that required I go back to 0.8.5 if I wanted to put it on my ESP-01.  Currently running fine on ESP-01 now with 0.8.5 but I did want to use DMX in the future so if that bug is fixed in future builds I'll hopefully find out by following that other issue's thread.

Thanks for the prompt reply and sorry for bothering you.

Aircoookie, 2020-08-03:
No worries, I hope the next release will work fine for you!

davidboon78, 2020-08-03:
Me too!  the 0.10 release looks dope!  I'll try it out on some NodeMCUs i have.