wled/WLED #590: Unable to create segments

exeljb, 2020-01-13:
Hi, so I feel like I’m not doing something right when trying to make new segments. I set a color and effect with segment 0 checked and change my stop LED. When I tap on Add New Segment, Segment 1 (new) shows up but no checkbox is there nor another Add New Segment button. What am I missing?  This seems like something that should be pretty straightforward. I’ve tried 3 different browsers on my Mac and both safari and wled app on my iPad and iPhones.

raoulteeuwen, 2020-01-14:
Do you see a check mark in the segment box? What happens when you press that? (for me, that adds another segment ... i agree maybe a + would be clearer compared to a check mark)

Def3nder, 2020-01-14:
Hi @exeljb , after pressing "add segment" you have to check/update the start/stop LEDs and then press the ✔️ 

After that you get the "add segment" button again and can add more segments.

exeljb, 2020-01-15:
Ahh the checkmark!  I was clicking the checkbox til i was blue in the face, thanks for your help folks!  I don't know why i didn't think to click the actual checkmark, i feel a bit foolish for not thinking of clicking that little guy.  A plus sign or a next arrow would've been a bit easier, or maybe the checkmark being located at the bottom right of the open segment window would've made it more obvious for blind folks like myself.  Thanks for the help!

Def3nder, 2020-01-16:
...if all is okay you can close the issue 😄

closed 2020-01-17 by exeljb