wled/WLED #593: Feature Request - Roll the Dice | Random Color, Effects, Etc.

josephdyland, 2020-01-14:
I have a Feature Request I think would be useful, some randomization like a Roll the Dice button and have different and selectable things be randomized.  I Drew and attached a Mock-up [image]
 of what it could look like.

Since it is so difficult to make up ones mind on which effect / colors to use. It would be great to be able to enable randomness on;

Primary Color
Secondary Color
Tertiary Color
Effect
Palette
Speed
Intensity

huggy-d1, 2020-01-15:
To expand on this excellent concept, I wonder if a subset of all effects / colors could be selected, like checkbox for "All" (original suggestion).   Suppose either a visitor or resident has epilepsy and we want to exclude all strobe effects, and limit speed to sub-seizure rates.

If "All" is unchecked, display effects with a checkbox (or other selector).

If "Favorite" is unchecked for color palette, then all colors.  If it's checked, only allow color randomizing from "Favorites" colors.

As an interim solution, maybe create a new effect with additional properties, named "Random effect", and new color, named "Random color".  When those are selected, we get the effect as if we rolled the dice, but just one time.  Once the dice is rolled (random function called) for each option, that option is now _selected_ as if we manually selected it.

The OP has an intriguing UI perspective where dice are rolled.  WLED presents new effect (and/or new color).  We look at our significant other's expression.  Depending, perhaps we keep rolling the dice, hoping to stumble upon an effect that suits our (or their) mood.  

I think this suggestion can increase WAF!  Probably fun to play with during parties too.

If I were to implement it, I would have Di shown in the UI, and when "roll the dice" button is selected, I would use JS to animate the Di and alter the # of visible pips on the face.  Either that, or find a free Di animation and activate it when the Di are rolled !!

sansillusion, 2020-01-15:
I support that idea ! That's what I am missing from wled.

Aircoookie, 2020-01-16:
I like this suggestion! A simple random color selector will in fact be part of the upcoming 0.9.1 version.

Randomizing effects and their parameters really sounds like a fun concept! Excluding some effects from the list or constraining the range of possible values for speed and intensity should also be done, I agree!

pniewiadowski, 2020-03-21:
Do we have an volunteer to take care of this? I really appreciate the idea!

blazoncek, 2022-10-10:
It is possible to use "r" as a parameter for "sx", "ix", "pal", "fx" (and others) keys in JSON API allowing for random values to be applied.
https://kno.wled.ge/interfaces/json-api/#contents-of-the-segment-object

closed 2022-10-10 by blazoncek