wled/WLED #594: Writing Lifx usermod; questions...

stringandstickytape, 2020-01-14:
Hi

I've been working on a usermod to add Lifx support to WLED : 

https://github.com/stringandstickytape/WLED/tree/LifxSupport

I think it's working pretty-well - but it's probably not quite ready yet for a pull request.  If it's not too cheeky, could you give it a once-over and see if you have any views?  There are fairly extensive notes at the top of wled06 about its capabilities and requirements.

Two particular points of interest (i.e. areas where it might be dubious):

* To allow WLED06 to control individual LEDs, I added a new FX_MODE_USERMOD which does nothing but stop WLED from doing any LED-setting itself.  Is there a better way?

* For Lifx purposes, I divide the strip into eight or 64 "zones" - Lifx requires multiples of eight I believe.  I think this has no overlap with your segments concept and should indeed be implemented separately as it currently is, but frankly I am not certain...?

Views greatly appreciated!

Aircoookie, 2020-01-16:
Hi,
that's sounds like a very interesting usermod and great work so far!

To take over control and stop WLED from rendering built-in effects, an effect that does nothing is a workaround, but there is in fact a better way (which is also used by e.g. E1.31)! Call `arlsLock(65000)` when you want the usermod to take over control and `arlsLock(0)` once you want WLED effects to run again. You can also specify a different number, in which case normal WLED effects would resume after that number of milliseconds (`arlsLock(2000)` would stop the WLED effect for 2 seconds)

The zones sound fine to me! Indeed this is separate from segments. Using `strip.setPixelColor()` will output to the raw pixel index you want (exept for the "Reverse entire strip" and "Skip first LED" settings, those are always honored)

Looking forward to your usermod and let me know if you'd like help with anything else :)

stringandstickytape, 2020-01-17:
Many thanks for your comments.  I *knew* there had to be something like `arlsLock(65000)` , saw it but didn't realise its significance at all.  This will simplify mod installation substantially so thanks for that!

egrim, 2020-03-14:
This looks great - how's it coming?  Need any help?

closed 2022-10-21 by Aircoookie

MaxVonEvil, 2023-08-13:
Hi, am looking around for exactly the same. I have a set of LIFX Z strips, but the controller is DOA. So what's the current state of things? Looks like not much has happened to that other repo the last 3+ years? Did WLED get support for Lifx strips in the meantime? 

Btw: Are the LIFX strips very different from regular W2812 strips? Any tips on how to wire up the lifx strip, i.e. what's pwr, gnd and data (my strips have no markings and the controller is obviously no help) And what's the 4th pin on the strip for?

Thank you!

yuuzhan, 2024-03-28:
In case anyone did decide to work on this, this is the circuit board in the controller.  

Back of the Chip:
[image]

Front of Chip
[image]

I think I am going to try to just wire it up to 5v and 24v as if it was a 3 pin strip and see what happens.

bamnet, 2024-08-31:
@yuuzhan did you have any luck with that approach?  I have a dead controller and am debating the best way to salvage the lights if possible.

yuuzhan, 2024-09-01:
Havent tried yet.  Was going to then realized I didnt have the right equipment then life got busy.  Still on my todo list in the next few months :P