wled/WLED #595: HW or software?

Anothermetrial, 2020-01-15:
Hi guys,
I installed 2 Wled NodeMcu (0.8.6) and 2 led strips on my house. Set up automations in HA to work together as in DR Zzss's repo. There are 2 issues:
1 - approx 20 minutes 1 strip (the long one) is turning off without any command from HA. Tried linking this to a specific effect but couldn't find a reason
2 - automation is turning the led show off at 22.00 and in the morning I have few leds in random colors lighten up continuously.

The scheme is:
[image]

The picture for 2nd issue.
[image]

Please advise where to start debugging and how. HW issue? SW issue? Thank you for your help

underwoodblog, 2020-01-15:
I would exclude a hardware issue first. I think the diameter of the cables is a little thin for this amount of LEDs and this cable length. Have you checked the voltage on the LEDs at full load?

If you're using the power supply on your picture, check if it has a trimmer to adjust the output voltage and set it to 5,1V at full load.

A big capacitor nearby every ESP would be a go idea to. Some Ferrite around the data cable may help to (multiple windings!).

How long are the data cables?

Anothermetrial, 2020-01-15:
Capacitor 3300 uF, 16 V is already mounted, same result. Data wire for the longest segment is 11m as in the picture.
I am using the leds at 20% brightness - enough for the desired holiday effects. Will check the last led voltage and the trimmer and come back with the results. Thank you

pepeEL, 2020-01-15:
Also try flash firmware in you nodeMCU. I also has nodeMCU that he has a problem that after some days WLED no respond on command. After flash by new FW all works ok.

Anothermetrial, 2020-01-15:
> Also try flash firmware in you nodeMCU. I also has nodeMCU that he has a problem that after some days WLED no respond on command. After flash by new FW all works ok.
> BTW What model of led you use on outdoor ? Can you give ma link ?

I will flash an EP32 with Wled and replace the "trouble" one.

I use WS2812 30 led/m IP65 in black aluminium channel https://www.aliexpress.com/item/32956009458.html?spm=a2g0s.9042311.0.0.68114c4d840SNT

Anothermetrial, 2020-01-15:
Arrived home and the 2nd strip has 1 green pixel on ....
[image]
I really don't understand.

huggy-d1, 2020-01-15:
Please describe where the power supply, NodeMCU, and level shifter is located for each strip.

To help isolate the problem, can you try the following:
Restart everything on strip #1 and immediately surround the nodeMCU with wax paper and then tin-foil (wax paper so you don't short out the nodeMCU).

When I say restart, I mean get it doing what you want which includes commands from HA as required.

See how long this runs.  On the 2nd strip, do the same with wax paper / tin-foil for the level shifter and leave the nodeMCU as you have it now.

With your 2 WLED setup, you can simultaneously troubleshoot 2 potential problems / solutions.

pepeEL, 2020-01-15:
Please show me also black aluminium channel

Anothermetrial, 2020-01-15:
> Please show me also black aluminium channel

https://m.aliexpress.com/item/32438887302.html?pid=808_0000_0131&spm=a2g0n.search-amp.list.32438887302&aff_trace_key=&aff_platform=msite&m_page_id=83013AeoC69yWIw7glBnKwFd0ReWfuaaPMql3MZhQmwf-hyLJJGD1lH9hE4uzBx2rlzX1579118872517&browser_id=91cfa51e3791475ea1cec94d9c80479c&is_c=Y

raoulteeuwen, 2020-01-15:
@Anothermetrial did you ever watch https://youtu.be/ZFO_QOBG9Bs?t=290 ? Maybe that episode contains info that helps.

mjferro, 2020-01-15:
For number 2, the random LEDs turning on is come form random noise on the data line.

When WLED turns off the LED strip, It does not continuously send an off command. So, if noise turns an LED on after the off command, it will come on and stay on. 

Its would be better to turn off the power that is supplied to the strip using a relay, then just relying on WLED. 

Also, I have found that using strip LEDs IP65 in aluminium channels like yours are horribly unreliable. I have a similar setup have been having nothing but nonstop moisture problems. I think water is getting between the double stick tape and the strip, where the exposed contacts are, causing corruption on the data line. My plan is to replace them with IP67 next year. So that may be something you should look out for.

underwoodblog, 2020-01-16:
Ferrite and some shielding can maby help for random turning on some led.

Isn't it possible to add some pullup or pulldown resistor to the data line?

Anothermetrial, 2020-01-20:
> Ferrite and some shielding can maby help for random turning on some led.
> 
> Isn't it possible to add some pullup or pulldown resistor to the data line?

What should be the resistor value? And where to mount it (between nodemcu and level shifter or after the level shifter)? Thank you

thedannymullen, 2020-01-21:
Looking at the first circuit diagram you have.  If you wish to have a level shifter I would use a null pixel.  In my opinion they are more reliable. 
What you are experiencing is noise related so you may have poor grounding from the node mcu.  I had to run two grounds on one of my strings one from the supply and one from the node mcu.  Connecting them at the strip helped.  

Lastly I would also put a sonoff in-line with power supply and turn power off when lights not running.

underwoodblog, 2020-01-23:
> Ferrite and some shielding can maby help for random turning on some led.
> 
> Isn't it possible to add some pullup or pulldown resistor to the data line?

The video that @raoulteeuwen mentioned has a better solution as a pullup or pulldown to make the data line stable. A single led as amplifier may be the solution. Putting some ferrite  on the wires is never wrong.

Anothermetrial, 2020-01-23:
I did already moved the power supply away from the level shifters&nodemcu, capacitor installed and still no improvement.
I will replace today one level shifter with 1 led according to your recommandation and come back with the result.

closed 2020-05-29 by stale[bot]