wled/WLED #596: Feature request: Password protection of adding lights to app

fribse, 2020-01-16:
Hi Aircookie

The 0.9.0-b1 is performing very well here. Very smooth.
I had an idea, because I've now added some LED's to my youngest desk, and as some features are not controllable through the IKEA remote I've given him, he has to have the WLED app installed.
This in itself is not a problem, everybody can just download it, but it is a problem that he (and others) can add our lights to their app, and create chaos for us (if they have access to our Wifi of course).

Thus I would very much like to have a password protection or the like, built in to the code, so that you have to give the password in the app to be allowed to add it.

This will eliminate the problem of him (and others) making funny effects with the other wled's in the house.

My thought was that you could set the password during the initial setup, and then to be able to add it to the app, you would have to enter the password as well.

Aircoookie, 2020-01-21:
Hi!
I am glad that you like WLED!

It would probably be a good idea for me to implement one of the following at some point:
- Password protection for UI
- API key system so that you need to approve any new client device before it may control the light
- Limited accessibilty, e.g. only from certain white-listed IPs

What you could do for now though, is after adding the light to every app you want to control it from, clear the mDNS field in WiFi settings. This will make it impossible for the app to discover the light (you can still add it manually by IP though)

fribse, 2020-01-22:
An API key system would be perfect, but might also be overkill :-)
I guess White-listing and PW protection would be the low-hanging fruits.
Ok, good to know. I'll do that to start with, as the other lights are already present, and connected to HA.

blazoncek, 2022-07-12:
#2610 has a PIN code as a protection mechanism for changing Settings.

blazoncek, 2024-07-26:
Closing this issue as it is directed to the WLED mobile app which has its own repository (unfortunately abandoned at the time of writing).
There is alternative mobile app by @Moustachauve with some advanced features.

closed 2024-07-26 by blazoncek

Moustachauve, 2024-07-26:
I feel like having a pin for settings was a good first step, but it seems like the real request for this bug would be to have the API and user interface be password protected so that someone with a "rogue" app can't start turning on and off all their lights.

Even if I wanted to implement this in the app, I would need a way to ask the controller if I have the correct pin or something like that. 

I don't know if it's a feature we want to have though. I can see how it can be useful for some cases, but it would add quite some complexity.