wled/WLED #597: Feature Request: Return current preset with http api response

sogeniusio, 2020-01-16:
Any way we can expose the current preset to the xml response when you hit 

http://123.456.78.99/win&PL=1

Def3nder, 2020-01-16:
Hi @sogeniusio ,
I've written such a thing for myself just to be able to kind of backup the presets.

I've changed wled02_xlm.ino to display another line with the JSON API:
LINE 80 changed from:
`oappend("</ss></vs>");`
to:
```
  oappend("</ss>");
  // append a string for use in html-API
  char s[16];
  oappend("<presetstring>");
  oappend("http&#58;&#47;&#47;");
  IPAddress localIP = WiFi.localIP();
  sprintf(s, "%d.%d.%d.%d", localIP[0], localIP[1], localIP[2], localIP[3]);
  oappend(s);
  oappend("&#47;win&#38;A&#61;");
  oappendi(bri);
  oappend("&#38;CL&#61;h");
  for (int i = 0; i < 3; i++)
  {
   sprintf(s,"%02X", col[i]);
   oappend(s); 
  }
  oappend("&#38;C2&#61;h");
  for (int i = 0; i < 3; i++)
  {
   sprintf(s,"%02X", colSec[i]);
   oappend(s);
  }
  oappend("&#38;FX&#61;");
  oappendi(effectCurrent);
  oappend("&#38;SX&#61;");
  oappendi(effectSpeed);
  oappend("&#38;IX&#61;");
  oappendi(effectIntensity);
  oappend("&#38;FP&#61;");
  oappendi(effectPalette);
  oappend("</presetstring>");
  oappend("</vs>");
```
This displayes an URL that can be copied to your browser and does exactly the same as loading the effect (sets the brightness, colors, effect number, speed, intensity and effect palette).

Would that go in that direction what you think off ?

Aircoookie, 2020-01-17:
@sogeniusio Hi, but in fact this is already the case. If you load a preset with the `PL=` command, the returned XML values for brightness, colors and effect parameters correspond to the preset you just loaded :)

@Def3nder I do not believe this is a terrible idea, however the API is meant for machine to machine communication and the information it adds is redundant. Of course it makes it easy to recall the state by just copy pasting in the command, but that's what presets are for :)

sogeniusio, 2020-01-17:
@Aircoookie - My use case is for Home Assistant. I want to use an input select to choose preset 1-16 via the api and a means for tracking the state.

Aircoookie, 2020-01-17:
Ah ok, now I get you :) What you want is a field that contains the number of the currently selected preset, not brightness etc. Sure, will add that!

sogeniusio, 2020-01-17:
Wow! Thanks.

-- 

Hosnel Guerrier
Founder of So Genius I/O

Aircoookie, 2020-01-19:
Done!

commit 125e2190 (2020-01-19): Added new HTTP API commands (resolves #597)
closed 2020-01-19 by Aircoookie