wled/WLED #598: 5 Channel Support

chaptergy, 2020-01-16:
I would like to control an RGBWW lamp with WLED. In the wikis FAQ it says 6 _pins: RGBCT (RGB + 2 white channels)_ is supported. However, when trying to use it by defining `WLED_USE_ANALOG_LEDS`, `WLED_USE_5CH` and apparently also `WLED_USE_H801` it does not work. Apart from a few typos like the [`#define WLED_USE_5CH`](https://github.com/Aircoookie/WLED/blob/cf60eb52d82fff10d67d4e014ba20ba568a85259/wled00/NpbWrapper.h#L12) having to be `#define WLED_USE_5CH_LED` and two instances[ of it saying `colorW` instead of `color`](https://github.com/Aircoookie/WLED/blob/cf60eb52d82fff10d67d4e014ba20ba568a85259/wled00/NpbWrapper.h#L259) it is still not able to control the second white channel. When looking though the code it seems, that the frontend doesn't support 5 channels after all, but even when connecting it to home assistant via the wled integration, the second white channel is unused (The pin stays at 0V). Is this just a mistake in the FAQ and 5 channels are not supported or am I doing something wrong?

Thanks

Def3nder, 2020-01-16:
Hi @chaptergy ,

the 5th channel is hard coded for 5 distinct color temperaatures that come from Alexa.

In EspalexaDevice.cpp lines 125ff the colors are encoded from CT to RGBW:
```
// Cold white to warm white receiving from Alexa: _ct = 199, 234, 284, 350, 383 (from cold white to warm white)
    switch (_ct) {
      case 199: rgb[0]=255,rgb[1]=255,rgb[2]=255;rgb[3]=255;break;
      case 234: rgb[0]=127,rgb[1]=127,rgb[2]=127;rgb[3]=255;break;
      case 284: rgb[0]=0,rgb[1]=0,rgb[2]=0;rgb[3]=255;break;
      case 350: rgb[0]=130,rgb[1]=90,rgb[2]=0;rgb[3]=255;break;
      case 383: rgb[0]=255,rgb[1]=153,rgb[2]=0;rgb[3]=255;break;
```

This works fine for 4-channel LEDs.
If there are 5 Channels, these 5 Color-combinations are quried and translated to pure warm + cold whites:
```
#ifdef WLED_USE_5CH_LEDS
            if        (color.R == 255 & color.G == 255 && color.B == 255 && color.W == 255) {  
              SetRgbwPwm(0, 0, 0,                  0, color.W * b / 255);
            } else if (color.R == 127 & color.G == 127 && color.B == 127 && color.W == 255) {  
              SetRgbwPwm(0, 0, 0, color.W * b / 512, color.W * b / 255);
            } else if (color.R ==   0 & color.G ==   0 && color.B ==   0 && color.W == 255) {  
              SetRgbwPwm(0, 0, 0, color.W * b / 255,                  0);
            } else if (color.R == 130 & color.G ==  90 && color.B ==   0 && color.W == 255) {  
              SetRgbwPwm(0, 0, 0, color.W * b / 255, color.W * b / 512);
            } else if (color.R == 255 & color.G == 153 && color.B ==   0 && color.W == 255) {  
              SetRgbwPwm(0, 0, 0, color.W * b / 255,                  0);
            } else {  // not only white colors
              SetRgbwPwm(color.R * b / 255, color.G * b / 255, color.B * b / 255, color.W * b / 255);
            }
#else
```

For all other colors the 5th channel is ignored.

You are right, there isn't a "real" translation from RGB to RGB-CT.

Could you pls. try if the values above do switch-ON the 5th channel ?

e.g. with `http://WLED-IP/win&R=127&G=127&B=127&W=255`
Does this work ?

chaptergy, 2020-01-18:
Hey @Def3nder,

Yes, those values do indeed work, however then I presume that it is not possible to control the last channel independently of the others? I'm trying to get the color temperature slider in the Home Assistant integration to work. Is WLED not the right option for this then?

Thanks!

Def3nder, 2020-01-20:
Hi @chaptergy , this is a good idea - I will look into it and see what can be implemented.

The original NeoPixel Library does only support RGBW and so the 5 channel support was added in order to get correct colors when using Alexa (and alexa only has 5 different color temperatures).

The whole WLED project runs with 32bit for the colors and therefore has R,G,B and W.
My workaround was to read this 4 values and create 5 ones (with warm white and cold white) if there is a special RGBW-combination set in the EspalexaDevice library.

So, to solve this WLED would need to switch from 32 bit color to 48 bit color.

We need to loop-in @Aircoookie to see if there is something that speakes against having in general an array of 5 colors.

chaptergy, 2020-01-20:
Awesome! I would love to help if there is anything I could do.

Def3nder, 2020-01-21:
Hi @chaptergy ,

I digged into the WLED home-assistant integration and it turns out that this does send only a 4-channel color (R,G,B and W) - so from this we cannot ever get a valid CT color.

From my point of view we would need to get @Frenck and @Aircoookie together and discuss what kind of "color-space" would be needed (in both HA and WLED) to get 5-channel RGB strips working.

Normally the 5-channel strips display either an RGB value or a CT value (a mixture of the WW and the CW channel).

I will support implementing the code, but without the "definition of the to-be-used colors" this is difficult and a lot of back-and-forth...

@frenck @Aircoookie : 

- What color array would be suited best for this ?
- Does WLED need to define the "features" so that the home-assistant WLED-integration does know what the light supports ?

frenck, 2020-01-21:
From my endpoint (Home Assistant), I'm happy to adjust to the API changes. I think considering WLED and the integration both, adding another element to the color tuples will do the trick (R, G, B, W, W), and of course, a flag to indicate the device supports RGBWW.

Def3nder, 2020-01-21:
So if we introduce a color array RGB[5] with {Red, Green, Blue, [White|CT], color-type] 
with color-type:
1 = RGB
2 = RGB with Warm White (e.g. 2700 Kelvin)
3 = RGB with Cold White (e.g. 4000 Kelvin)
4 = RGB with Natural White (e.g. 6500 Kelvin)
5 = RGB + CT (color temperature from 2700 to 6500 Kelvin)

We could further add fields in the settings to enter the kelvin values of the different W-channels.
And the API would need to accept another entity "CT" with a kelvin value for the white color tone.

Does this make sense ?

Aircoookie, 2020-01-21:
@Def3nder I believe adding a 5th channel (and flags indicating the particular device supports it) shouldn't be too involved.
It may be a problem though as far as effects are concerned, as they use `uint32_t` all over the place. That would have to be substituted for a new color object supporting more channels in any function that deals with them (exept the FastLED palette stuff, it doesn't even support a single White channel).
It would make the color processing slower as well as use more memory, though we'd need to test how much real-world inpact it would have.
I agree though that a better white color management would be a very nice feature to have.
This could also be interesting for controlling these WarmWhite - ColdWhite - Amber strips with a single slider.

@frenck what do you reckon, would you rather do color calculations on the HA side and send 3, 4 or 5 channels depending on the flag(s) returned (as we are doing now with the white channel), or alternatively, return a kelvin or mired value and I will try to map that to RGB, RGBW (with different color temperatures), and RGBWWCW inside WLED?

frenck, 2020-01-21:
@aircoookie, from an API perspective, sending in just RGB or color temperature would be awesome. Please note, keeping some backward compatibility would be appreciated ;)

Aircoookie, 2020-01-22:
@frenck Alright, I also figured that would make the most sense, because it enables the CT capability to also work via the web UI and other non-HA clients.
For backward compatibility, new versions will of couse still support raw channel input. As for native color temperature capability, you'd just need to check that the version ID (`info.vid`) field is at least the one where I'll introduce the support, or I could optionally add a field indicating CT support, though even RGB lights would support it :)

Off-topic, but some users said they'd like to see signal strength added to the WLED sensor. In 0.9.0, `info.wifi.signal` returns a relative quality, maybe we should wait until the release of 0.9.1 though, since that also supports `info.wifi.rssi` (added in ef4ec16860654e7d62e7bca89d2c9436118fb12e), a more standard metric that is also useg by e.g. the ESPHome sensor (see https://github.com/Aircoookie/WLED/issues/434#issuecomment-570950082)

frenck, 2020-01-22:
@Aircoookie The can build the vid check into the WLED Python library, that way, even Home Assistant doesn't need to know :)

About the WiFi signal, yeah, I would love to add that. For the upcoming release (Home Assistant 0.105), I've added the ability to disable/enable entities (sensors/lights/switches/segments) as a user. E.g., in case the nightlight functionality is not what you need for that specific light.

Furthermore, a needed change to Home Assistant has been made, now allowing me to start adding in additional things like speed control, palettes, etc. So that currently has my priority.

Def3nder, 2020-01-22:
Hi @Aircoookie ,

I think we do not need to change the unit32_t for the color values of the effects.
The only constrain would be that in effects we only deal with 4 channels (or even with RGB only).
But this should be okay - the "commercial lights" that I know treat the CT that they show either an RGB color or a CT white.

I would suggest to have an array like `color[] ={byte W, byte R, byte G, byte B}` for the effects and a second auxiliary array `colorFeatures[] = {byte W2, byte CT, byte features}`
with W2 = value of cold white, CT = color temperature in mired and features that define what the light can display (like "RGB", "RGBW", "RGBCT").

Then we would need additional settings for the CT of both white LEDs.

With this we could calculate the right combination of white tones when a color "comes in" from either EspalexaDevice, from the API, from the UI or via MQTT.

tonyn0, 2020-01-25:
I didn't see anyone addressing the apparent typo mentioned: WLED_USE_5CH defined, but WLED_USE_5CH_LEDS used. ;)

Def3nder, 2020-01-27:
Hi @tonyn0, thank you for this info - I will include this in the next PR for this topic.

Aircoookie, 2020-01-27:
Fixed the `WLED_USE_5CH_LEDS` in dc936b63d6f6fb92df71d8e3b1ab7b214506f9bf :)

harueg, 2020-02-29:
Hi
Sounds wonderful !
I would like to add a thought to it. 
For a real lamp, real in the meaning of for everyday use., it would be great to have a slider for the brightness of white and another one for the CCT. The whites would be on a separate strip, with a high CRI. From my experience the CRI of the white on the multicolor addressable chips is quite abysmal.
The RGB would be just for additional effects.
Anyway, which chip supports 5 colors ?

Basti149, 2020-07-09:
Ich habe gelesen das es auch möglich ist mit einem ESP8266 und normalen RGB Strips man auch WLED nutzen kann. Gibt es eine fertig angepasste bin Firmware da ich selbst noch nie irgendeine Firmware geändert habe. Mit der man bei Wled auch 4 pin RGB LEDs nutzen kann. Oder ist es bereits in der aktuellen Firmware integriert?

rlsmntr, 2021-01-04:
This still shows as open but hasn't seen any movement in a year.  Is this still being worked on?  I pulled the latest source and it does not appear in it.

rjgrandy, 2021-01-31:
Also curious on the status of this. I've been trying to get the 5 channel working all night, as it states there is support but no dice so far...

megawusl, 2021-02-21:
Same here. After some hours of trial-and-error with 5 channels I've found this issue which at least tells me that I am not stupid. An update on this topic would be great.

roboraptor, 2021-02-22:
Can somebody provide *.bin with working analog 5CH leds support on ESP8266 please? 😅 

I don't think I did something wrong (I followed the manual) but only red channel works for me... 😭

megawusl, 2021-02-22:
> Can somebody provide *.bin with working analog 5CH leds support on ESP8266 please? 
> 
> I don't think I did something wrong (I followed the manual) but only red channel works for me...

need to know your board and the pin you want to use for each color. then I could compile a bin.

rjgrandy, 2021-02-22:
Can you really make one that works for 5 channels? I was under the impression this wasn't possible with the current software. Only 4 channel support.

megawusl, 2021-02-22:
> 
> 
> Can you really make one that works for 5 channels? I was under the impression this wasn't possible with the current software. Only 4 channel support.

you are right. Only 4 channels are actually working - the fifth is not. Understood the request from roboraptor in a way that he wants to get a 5-channel  *.bin (at least working for more than only red). Still - I can do it. Got 4 channels working for me and I am at least satisfied as I could test my developed board design.

roboraptor, 2021-02-22:
> need to know your board and the pin you want to use for each color. then I could compile a bin.

Hi, Thank you for your quick response and being helpful! 😄  

My board is plain ESP8266 connected to wifi controlled by WebUI and Alexa.
(At the time with Tasmota and all works good, just without any effects 😢 )

That EGLO Controller/Ballast needs pwm for:

- Red
- Green
- Blue
- Cold white
- Warm white

Is feature like only RGB or only CW white avaliable? (Power supply is not balanced to power all at once)

roboraptor, 2021-02-24:
> you are right. Only 4 channels are actually working - the fifth is not. Understood the request from roboraptor in a way that he wants to get a 5-channel *.bin (at least working for more than only [...]

Yeah, I gues I can split the 4th PWM signal for both cold and warm whites 😄

megawusl, 2021-02-24:
Hi roboraptor,
try this bin. This is working for me (only 4 channels). You have to unzip first and then you can upload the *.bin
Colors to Pin combination as the following:
Color >> Pin Description = Pin ID for WLED
red   >> D5 = 14
warm white >> D6 = 12
green >> D7 = 13
cold white  >> D2 = 4
blue  >> D4 = 2
remember, only 4 channels really work. In my case the cold white does not work. 

I've set a low PWM frequency of 340 Hz to test the reason for some nasty noise I have on my PCB. Thus, if you recognize LED flickering this might be not a bug but a special feature of my build :) 
 
[d1mini_5CH_H801.bin.gz](https://github.com/Aircoookie/WLED/files/6038361/d1mini_5CH_H801.bin.gz)

megawusl, 2021-02-24:
> Yeah, I gues I can split the 4th PWM signal for both cold and warm whites 😄

Btw., good idea. If your mosfets are strong enough and stay cool this will work. Just color temperature will not be perfect. I will try this too

roboraptor, 2021-02-25:
> Btw., good idea. If your mosfets are strong enough and stay cool this will work. Just color temperature will not be perfect. I will try this too

Only thing is, Tasmota turns off rgb when I fiddle with/turn on white, and turn off white when I fiddle with/turn on rgb.
Is this "featerue" here too? Because original power supply is not capable of running both on full power.

megawusl, 2021-02-25:
don't think this works. RGB is used to simulate white tones. As far as I could find in the code this seems to be the case for all RGBW_MODE_* options. Thus, you either need to cut your LED strip to or need a bigger power supply and I suggest also a big capacitor to support the switching of the fets ;)

blazoncek, 2021-11-29:
Will be fixed by #2285

closed 2021-11-29 by blazoncek