wled/WLED #605: Presets not saving and rebooting

Aphris-Karu, 2020-01-19:
I compiled from the current source and installed into a nodemcu with 4m flash with 1m SPIFFS. I also had it clear the flash during install. Im running 300 leds broken into 5 segments of 60 leds (kind of matrix layout)

Im seeing it not want to save favorites. It is not every time, I can do it several times and it not take then save it and it suddenly takes. 

I have also noted it rebooting from time to time, not sure what is up with that. I should say, I think it is rebooting as it stops responding, the app tells me it can not connect, and then when it comes back all the segments are gone until I load preset 16.

recliq, 2020-01-20:
I'm not quite sure what your problem is, because you don't tell what exactly you are saving to which preset slot...

Keep in mind that only slot 16 can save multiple segments!

You can also configure in LED settings to apply preset slot 16 on startup.

Since this an reactive webinterface it might also be a network problem... check network connectivity to your WLED while saving presets (eg. ping it)

Aphris-Karu, 2020-01-20:
Sorry, It is when setting presets that it fails to save. (Fixed in the above) I am aware that slot 16 is the only one that can save the segments. Though I did not think about loading preset 16 at startup. Ill have to set that up. Im still testing to see if I can ID a condition change that makes the difference between saving and not.

The reboots appear when using the interface at random points. I will be setting a color or pallet or effect making some change and the interface will suddenly stop responding and the unit reboots.

Side note: Very strange things happen when you delete segment 0 while having multiple segments.

Def3nder, 2020-01-20:
Hi @Aphris-Karu , most common if a NoceMCU is rebooting this is either a power problem (voltage breaking down) or a component connected wrong (e.g. without ground etc.).

Another source for this are cheap ESP devices - sorry for that, but the cheap china modules often have bad quality. 

So, please check the power supply (measure the voltage at the NoceMCU when all LEDs are set to the maximum brightness and try (if you can) another ESP module from a different supplier.

Aphris-Karu, 2020-01-21:
@Def3nder Thanks for the info. It is a chinese Knockoff NodeMCU. I had 5 of them but this is the last one. The others are running on different projects and I have not seen any stability issues with them, but that does not mean this one is not an issue. I have 10 more on order 5 of the nodemcu and 5 of the wemos mini's. 
I do have 13 of the ESP8266-01's on hand, they are the black ones with 4Mbit chips on them.

Checking the voltage it appears to be fine so Ill try another nodemcu when they get in.

Aircoookie, 2020-01-21:
@Aphris-Karu sorry for the issues! What are your compile settings? I'd recommend Lwip 1.4 and Arduino core 2.6.2 or 2.6.3. Also, I would be interested in what those very strange things are that happen for you if you delete Segment 0 :)

Aphris-Karu, 2020-01-22:
@Aircoookie I set up seg 0 with 0-300, then seg 1-5 with 60 led increments starting with 0-60. When I removed (deleted, not unchecked) seg 0 and saved in slot 16, all segments disappeared and it threw an error even after reboot it threw the same error. When I recreated seg 0 all the segments came back.

I say it is not saving but I have been working on it some and it appears that it does save but parts of what is being saved do not. For example, I set color 1 to blue, color 2 to black, color 3 to green, then set the effect to sparkle and save in slot 8. When I reload it the color set is the yellow I use in slot 1 and initially it had the solid effect but after a few saves it finally took the sparkle effect but the color was stuck at the settings for slot 1. (Note: After setting up ntp and getting the time correct it did finally save in slot 8)

Someone asked if I could check the network. Pinging the unit and browsing the interface shows 2% packet loss with the longest response time being 90ms. (I have several new nodemcu's and wemos mini's on order so once they are in I can swap this one out and see if it is a hardware issue)

Im guessing that by Arduino Core you are referring to the Arduino board core for esp8266 which is 2.6.3 and not the arduino core which is 1.8.2 running in the 1.8.5 IDE.

Some things I have noticed that may or may not matter. 
1) turning on NTP and setting the time improved the units response time and performance. 
2) It seems to reboot and I am trying to nail down what is the cause, though it is acting like one of my bigger programs when I was running into heap issues on an esp8266. Sometimes it reboots and sometimes it has the wrong data. For example, I noticed the year suddenly jump to 2106 when trying to save a favorite. This could be flakey memory on the unit it self. So I am waiting on the replacements.
3) I noticed that it would not connect to my device network which use 128bit WEP, I had to connect it to my share/guest network with WPA. I did not see anything in the documentation about it not working with 128bit WEP.

Aphris-Karu, 2020-01-22:
These are mounted in a fluorescent light fixture on the ceiling. 

Pictures are [HERE](https://drive.google.com/open?id=1merNqI3a4LGpSSzh49v-A1AbLo_Qi-hx)

Def3nder, 2020-01-22:
Hi @Aphris-Karu justan idea that came to my mind as I saw your last pictures:

you can also attach all 5 stripes directly to D4 and treat them as one segment:
[image]

closed 2020-05-28 by stale[bot]

rivelta, 2022-05-07:
why wled its not saved on wemos when reboot