wled/WLED #606: WLED becomes unresponsive after turning the LED stripe on

Cludch, 2020-01-19:
Hey,

I just installed WLED for the first time on my NodeMCU (ESP8266) and set it up to connect to my home WiFi.
Afterwards I connected my WS281**3** to 5V, GND and the Data-IN to D4. Powering everything up, the first LED blinks orange and then the stripe powers off (as I defined in the settings). But as soon as I power up the whole led stripe via the web interface, the NodeMCU crashes, dies or so.

The power supply is a 12V 3A one, which I connected to a DC converter in order to convert to 5V.
And the stripe is a WS2813 5m60LED.

Any clues, what I am doing wrong?

raoulteeuwen, 2020-01-19:
5m60 makes 300 LEDs. 60 mA per LED: afaik, your strip would (at max, but nevertheless) need a 18A power supply.

I think you could try, while no strip is connected, configure in WLED you only have for instance 20 LED's, and when you power cycle the nodeMCU and you have verified that 20 LED's is still configured as a number, attach your strip and see if it can handle that. Not sure if this actually limits the power the strip needs.

Cludch, 2020-01-19:
Thanks for the very fast response!
Yes that did indeed to the trick.
I already thought that the power supply was too small, but I tried to limit the usage via the webinterface and nothing changed.

Thanks again :)

closed 2020-01-19 by Cludch