wled/WLED #609: Animation Hiccup/Stutter/Signal-Loss

chuyskywalker, 2020-01-21:
https://www.youtube.com/watch?v=9jUkSgjsjYA

_Video to show issue_

If I've left the NodeMCU device I have with WLED on for a day or so, the animation for the rainbow effect will "stutter" occasionally (you can see it happen twice in this video). If I reboot the controller the effect will go back to normal with no stuttering.

I am on version `0.9.0-b1`

My nodemcu is by [KeeYees](https://www.amazon.com/gp/product/B07HF44GBT)

sogeniusio, 2020-01-21:
How are you wired?

chuyskywalker, 2020-01-21:
[12v lights](https://www.amazon.com/gp/product/B01AG923EU/), 3 strings 3m long with a second +/- input at the 2-3 junction. The controller feeds off a [12->5v](https://www.amazon.com/gp/product/B00J3MHRNO/) step down, all shared on the same [12v 8amp power supply](https://www.amazon.com/gp/product/B07G5BQGYD/).

[image]

Def3nder, 2020-01-21:
Hi @chuyskywalker , did you try flash WLED version 0.8.6 ?
Does this happen with this version, too ?

chuyskywalker, 2020-01-21:
> did you try flash WLED version 0.8.6 ?

Nope, I've only started with WLED a month ago. 

Is there some way to remotely collect system metrics from the controller that could help diagnose this?

Def3nder, 2020-01-21:
Hi @chuyskywalker no, remotely you could only look at `http://WLED-IP-Address/json` 
You will get a JSON reply that can be most easily read with the new firefox browser.

There you could monitor the entity "freeheap" (included in the "info" entity.

Sometimes it's good to compile the program with IP stack:
default is LwIP2 and it's set in the Arduino IDE or the PlatformIO.ini:
`build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH`

I assume you did flash the bin file "WLED_0.9.0-b1_ESP8266.bin" from the releases page...
Pls. try the attached release - it's a few weeks newer and compiled with LwIP 1.4.

You can upload a new binary in the settings under "Security & Update"
[WLED 0.9.0b2 pre LwIP1.4.zip](https://github.com/Aircoookie/WLED/files/4092114/WLED.0.9.0b2.pre.LwIP1.4.zip)

chuyskywalker, 2020-01-22:
Alright, I've got another of the 3-pack flashed with the version linked above, I'll report back if that resolved my stutter. If it doesn't, I'll try reflashing the same nodemcu board that had the issue in the first place as well, just to be sure.

chuyskywalker, 2020-01-22:
Well, didn't even have to wait long, about 40 minutes later the stutter is around. Worth re-flashing the first board?

vwtune, 2020-01-22:
already tried another powersupply and/ or step down controller?

Def3nder, 2020-01-22:
@chuyskywalker would you mind flashing it with 0.8.6 ?

...just to see if this has been there in this release, too.

brentbrooks70, 2020-02-05:
I cannot see in your picture but are the 12v and 5v grounds bridged anywhere? Does the step down share grounds? If not, you will need to.

closed 2020-06-11 by stale[bot]