wled/WLED #610: JSON errors (0.9.0.B1)

vwtune, 2020-01-21:
Since i use version 0.9.0.B1, i get syntax errors when i use the sliders of FX and/ or speed control. It is not by every effect. So when i choose an effect and then i change the fx or effect speed i get: syntaxerror: unexpected end of jason input.

Any ideas to solves this?

Def3nder, 2020-01-21:
Hi @vwtune this could be a "low memory" problem.
The change of any slider does send a JSON request and this allocated memory.
Some of the effects consume a lot of memory and if you then change a slider a couple of times the ESP gets an out-of-memory error.

Is this with all effects ?
Is it the same if you only slowly change anything in the UI ?

vwtune, 2020-01-21:
Hi @Def3nder thanks for your reply, This is not with all of the effects, but lets say 50% of the effects.
I use Wemos and Nodemcu devices, and they have the same problems. On Version 0.8.6 i have no problems at all.
When i change slowly in the UI, for my feeling, less often.
What can i do about it? Is this not a common problem in version 0.9.0?

Def3nder, 2020-01-21:
let me say it this way: version 0.9 did add a lot of features and some of them allocate a lot of memory ...

So, this does result in problems that had not been there in 0.8.6.

But only @Aircoookie himself can tell "the truth" - this is only my understanding.

I do see some of these errors in the UI, but then I do the action again and it does work.
The ESPs do not reboot or do anything that renders them unusable, so it's kind of okay for me 😄

vwtune, 2020-01-21:
@Aircoookie will this problem solved bij an newer version?

Aircoookie, 2020-01-21:
@vwtune see my response on discord :) This is most likely fixed in current source and an upcoming release!

closed 2020-01-23 by vwtune