wled/WLED #611: esp12e support

sungla55guy, 2020-01-21:
support for generic esp12e chips.  I'm using the H803 https://www.aliexpress.com/i/32854807170.html. and it uses the esp8266 esp12e 4M chip.

Def3nder, 2020-01-21:
Hi @sungla55guy this should work but you need to find out to which ESP GPIOs DAT and CLK are connected. 

Do you want to drive an APA102 LED striup with this ?
If not you only need to find out the corresponding GPIO for the DAT connector.

If you are lucky then you get GPIO2 for DAT and GPIO0 for CLK.
Then the APA102 release (WLED_0.9.0-b1_ESP8266_apa102.bin) would work out-of-the-box.

sungla55guy, 2020-01-21:
DAT is GPIO13 and CLK is GPIO14 which I’m not using since I have ws2812. It also has 2 status led’s red at gpio5 and green at gpio4. Ive compiled myself and got this working just not positive I did it totally correct

Def3nder, 2020-01-21:
please try to flash the attached release - it's with LEDPIN assigned to 13 (in NpbWrapper.h).
[WLED_0.9.0-b1_ESP32_ledpin13.zip](https://github.com/Aircoookie/WLED/files/4092352/WLED_0.9.0-b1_ESP32_ledpin13.zip)

Def3nder, 2020-01-21:
sorry - did not read until the end ... 😄 

What do you mean with "got this working just not positive I did it totally correct" ?
Does the LED strip show the correct lights or not ?

srg74, 2020-01-22:
Your controller will work no problem as soon it's using same module as NodeMCU and Wemos D1 mini. All you have to do is compile your version of firmware with DATA pin and CLK pin accordingly. Once flashed everything should work fine.

imduffy15, 2020-02-10:
Thinking about purchasing one of these units as its a very nice enclosed solution. Have you folks managed to get WLED on it successfully? Anyone got photos of what it looks like with the case off? Is their headers or is soldering required?

srg74, 2020-02-10:
Looks like @sungla55guy managed to flash WLED in to this controller. Just pay attention which pin he is using.

sungla55guy, 2020-02-11:
The header pins are exposed similar to a sonoff. They aren’t completely labeled but the they go in order with the last one being gpio0. In platformio.ini I used board type esp12e. However I’ve been having frequent home assistant disconnects with the latest dev build

imduffy15, 2020-02-11:
Awesome! Thanks for sharing, glad to hear there are actual pins :)

srg74, 2020-02-13:
@sungla55guy 
I have similar issues with connection to HA only with boards with weak signal. And this is one of the reason why I did custom PCB with external antenna for outside applications.

fleshinachair, 2020-05-10:
Hi guys
Would one of you be able to share your compiled firmware for the H803 please? I've no idea how to do it. Many thanks!

srg74, 2020-05-10:
@fleshinachair You can download from here https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/0.10_2005030/ESP8266 and explanation is here https://github.com/srg74/WLED/wiki/H803WiFi

fleshinachair, 2020-05-10:
thankyou for taking the time to do this, I will give this a go!

imduffy15, 2020-06-15:
@srg74 Can you confirm the RX pin is the one circled in your photo at  https://github.com/srg74/WLED-wemos-shield/raw/master/resources/Images/H803FW.jpg rather than the trace where the arrow ends

Also, can you confirm that GPIO0 needs to go to ground on boot?

srg74, 2020-06-15:
@imduffy15 If your board is looks the same than yes pins are correct. I flash this board that way. And to put esp8266 to programming mode you have to pull gpio0 to ground.

imduffy15, 2020-06-15:
Works perfectly! Massive thank you @srg74 . Been looking for a suitable device with screw terminals for ages, not many out there. Works very well for a real life, permanent installation (kitchen lighting).

srg74, 2020-06-15:
> Works perfectly! Massive thank you @srg74 . Been looking for a suitable device with screw terminals for ages, not many out there. Works very well for a real life, permanent installation (kitchen lighting).
Not necessary true. I have a board for wemos and @quindor, plus my weather proof box. Look at GitHub

closed 2020-10-21 by stale[bot]

Gedankenleser, 2021-04-25:
Any idea how to get the onboard "LED" to work? (D1, D2)