wled/WLED #614: Support 32 ft 5050

neuman1812, 2020-01-22:
Hello,  not sure if this is related to #58 or not.  Does wled support this product?

https://www.amazon.com/dp/B0713ZTK8J/ref=cm_sw_r_cp_apa_i_uRlkEbKVZ7M30

I have the wifi controller with proper power supply,  i was able to flash the 1mb version of wled.  But all i can do is turn it on and off.  I believe its a ws2811.

srg74, 2020-01-22:
As soon your LED strip is analog all you will be able to do is to turn on your strip and change color on entire strip to the same color. I believe some patterns will work too.

Def3nder, 2020-01-23:
Hi @neuman1812, yes, this strip is supported and the enclosed IR remote control should work with `Infrared receiver type` set to `4` in the **Sync Interfaces** setup.

The release to use have to be compiled with `build_flags = -D WLED_USE_ANALOG_LEDS` - if you use **PlatformIO**, you can uncomment the line `env_default = esp8285_4CH_MagicHome` in the platformio.ini.
There are some different versions of the MagicHome controller floating around - please try this bin: 
[WLED_esp8285_4CH_MagicHome_pre0.9.b2.zip](https://github.com/Aircoookie/WLED/files/4102176/WLED_esp8285_4CH_MagicHome_pre0.9.b2.zip)
It's the 4CH_MagicHome version from the actual master branch - if this does not work we need to find out what GPIO pin your version does use ...

The constrain of using "**solid color LED strips**" is that you can only change the color of the whole strip at once. Most of the effects do not look well - some good ones are "breath", "candle" or "palette".

Def3nder, 2020-01-23:
...there is one thing that you need to pay attention to:
You should inject power at least every 15 ft and use a power cable that is at least AWG 18 ( 0,75mm² ).

closed 2020-05-29 by stale[bot]