wled/WLED #615: Home Assistant Native Integration - Slow to update changes - :Polling?

baremoney, 2020-01-23:
Hello, I am running a **home assistant** automation that reads the one of the attribute states and if does something if the **attribute state change** is true. The problem is the **native integration is slow** and can on average take anywhere from a couple of seconds to a few seconds to reflect the state change. While using **mqtt before**, the changes were **updated much faster as if instantly**. **Does the native integration use some kind of polling interval to reduce overhead and is there a work around or way to have mqtt publish the nighlight state?**

Def3nder, 2020-01-23:
Hi @baremoney, this is a question belonging to the logic used in home assistant.
@frenck was the one that wrote that awesome WLED integration. I cannot really read the python code, but it looks like using a timer (polling).

The WLED itself does only send MQTT notification on any change in the UI as far as I know.

So it looks like the WLED integration is using the JSON API rather than MQTT ?

frenck, 2020-01-23:
Yes, the WLED integration uses the native API and does not require or rely on MQTT. There is no streaming kinda endpoint available on WLED, which makes it a polling integration.

However, since the device is local, we use a relatively high polling rate. For most use cases, it should not matter that much. Changes from HA are send instantly and applied instantly by WLED and internally instantly reflected withing HA itself, however, if changed from an outside source, HA needs max 9 seconds to catch up.

Def3nder, 2020-01-23:
@baremoney, does this answer your question ? If you need the nightlight change even faster than this 9 seconds, you could create an MQTT sensor for this in HA. 

The problem with this is that the MQTT response does contain XLM and no JSON 😢 
(and an easy way to create a sensor from an MQTT topic does rely on JSON...

@Aircoookie: is there a reason why you used XML in the `/v` topic in MQTT ?
...could that be changed to deliver the JSON part that is in the `state` tree ?

baremoney, 2020-01-23:
@Def3nder @frenck Thanks for the helpful overview about the integration! Having looked at the xml mqtt published I see the entries:
`<ns>0</ns>
<nr>0</nr>
<nl>0</nl>
<nf>0</nf>
<nd>1</nd>
<nt>1</nt>`
The wiki says mqtt is being overhauled by another user so I guess they have added the timelight option i was looking for , ie, nl ?

Def3nder, 2020-01-23:
Hi @baremoney, 

the entries have the following meanings:
`nl: nightlightActive`
`nf: nightlightFade`  (0 = off, 1= on, 2= fading from actual color to the secondary color)
`nd: nightlightDelayMins` (in minutes)
`nt: nightlightTargetBri`

Def3nder, 2020-01-23:
I've just tested it and it does work - I've created PR #617 to add a compile switch for this.

closed 2020-05-29 by stale[bot]