wled/WLED #616: Support of matrix

wunderbaum, 2020-01-23:
Hi,
is there any plan to support such led arrays:

https://www.aliexpress.com/item/4000345359803.html

srg74, 2020-01-23:
Matrix is WS2812B so it's supported by project.

Def3nder, 2020-01-23:
@wunderbaum, you can define 16 segments with 16 pixel each, then the selected effect will show across the whole pannel.

Aircoookie, 2020-01-23:
Yes, matrixes are supported, but there are no effects yet that take advantage of them.
That is planned, but there are a few more features i'd like to add first.

16 segments are unfortunately not possible, 10 is the limit right now.

wunderbaum, 2020-01-23:
Ok, thanks for the quick response!
When I order now, it takes 6 weeks 
Is it worth it?

Aircoookie, 2020-01-24:
@wunderbaum those matrices definitely look cool even with most of the 1D WLED effects. If you are interested in E1.31 driving (e.g. Vixen, xLights), you can also use 2D effects. So I would say, go for one :)

yusufsulaiman, 2020-01-25:
this is my matrix https://www.youtube.com/watch?v=qjyDDL7gY1I&t=33s
thanks @Aircoookie

KaizenKuba, 2020-02-05:
> Yes, matrixes are supported, but there are no effects yet that take advantage of them.
> That is planned, but there are a few more features i'd like to add first.
> 
> 16 segments are unfortunately not possible, 10 is the limit right now.

I hope you will add support for LED matrix. since rocket life has ceased to exist you have a great angle to play.

tkerby, 2020-03-01:
I’d really like to see matrix support. I’ve got WLED running on the M5Stack Atom Matrix board which is a micro 5x5 led matrix and esp32 configuration (about an inch cube). The web interface is a little slow but would be very cool if some matrix effects were available.

alperkal, 2020-04-07:
hi @tkerby can you share how you got Wled running on the Atom Matrix? thanks

djasonic, 2020-08-23:
Another vote for matrix, and to suggest option for each row / column to have different length and base offset from center to enable layout of a matrix within any convex polygon and not just rectangles.  For some shapes such as triangle or parallelogram, ability to do a half 'pixel' offset for rows would be nice to have and still see fx display accurately.

16 sounds like a nice power of 2 for segments, if resources allow for it.

jlaitinen81, 2020-12-14:
Any news on matrix support? I'm a big fan of WLED and would live to see some 2D effects on my 16x16 led matrix.

pruchai, 2021-01-08:
@jlaitinen81 

Matrixes work out of the box. I have 3 x 32 x 8 WS2812s chained into 96x8 strip. It works great. 

What I am having issues with scrolling text. I can not get Jinx to produce any output whatsoever, QLC+ works, but produces ugly output and completely unreliable. Within 30 seconds some letters just flip upside down and start producing a tail that never goes away. I am not familiar with DMX on WLED and whatever help resources exist, they are rather cryptic, incomplete or outdated.

jneilliii, 2021-01-17:
> I can not get Jinx to produce any output whatsoever

I was having this issue to and found that using the output patch from the setup menu I could click the fast patch button and after doing that it would work.

viktor6, 2021-11-30:
How to make a 16x16 matrix and 2D effects work?

blazoncek, 2021-11-30:
Use soundreactive fork.

viktor6, 2021-11-30:
@blazoncek I did not understand you

blazoncek, 2021-11-30:
Vanilla WLED does not yet support matrices. Soundreactive fork does.
Use Discord for support questions. Link on the front page.

RBSVader, 2022-07-18:
So, maybe anoyne can help me  with 12 strips  8 pixel each (96 led total), zigzag connection? Do i need to make 12 segments with each 2nd with reverse order or there is special configuration?

blazoncek, 2022-07-18:
As of May 2022 I am experimenting with 2D implementation.
Alpha grade binaries are available from @srg74 in his Wemos Shield repository (check experimental folder).
2D support may be included in WLED 0.14 later this year.

RBSVader, 2022-07-18:
Thanks. Tried "sound reactive" 0.13.1 version ([here](https://github.com/atuline/WLED)), working well so far.

blazoncek, 2022-10-10:
0.14 will add 2D support for matrices.

closed 2022-10-10 by blazoncek