wled/WLED #621: MagicHome controller - some info on flashing but also looking for some help

bradmck, 2020-01-23:
Hi All - first, a big thank you to Aircookie for this awesome project!  I bought some Magic Home addressable LED controllers to see if I could get WLED on them.  I have both some useful information to share on flashing as well as a question on why I may be having problems with OTA update.

First the information:
I had seen others able to flash Tasmota/ESPurna/etc.  I immediately ran into problems flashing and thought I messed up the module.  I simply could not get it to flash.  I hooked up the wiring as needed (GND and GPIO0 to ground and RX/TX connected) and could see the Bootmode as 1, but it refused to flash no matter what I did.  Thinking I messed the ESP module up I used a heat gun to get it off the PCB.  I soldered leads to the back of it and it flashed first time.  I mounted it back on the PCB and it works with WLED fine as long as I jumper the LED data output to GPIO2 on the ESP.  I received a new module yesterday and had the exact same issue.

A couple of days ago I found a thread on Reddit about these exact modules.  Yesterday someone replied saying you need to remove a couple of resistors (or cut traces) to get this to flash.  It's because TX/RX are actually connected to an ARM chip on this board.  Removing the resistors breaks that connection.  I tried it and he's completely correct.  Explains why removing my first ESP from the PCB worked.  Flashed first time with no issue.  Here's the Reddit thread for anyone interested:  https://www.reddit.com/r/esp8266/comments/erz5zo/anyone_have_experience_flashing_one_of_these_or/

Now my question:
So I got the module to flash which is fantastic.  But I can't get OTA updating to work.  I thought maybe it was a memory issue so commented out all of the Blynk, Alexa, IR, etc stuff that I didn't need and flashed that over the wire.  Then compiled that to a BIN and tried to flash OTA and had the same issue.  It always comes back as:
Failed updating firmware!
Please check your file and retry!

This is an ESP-M2 so an ESP8285.  Is this a problem with lack of memory?  I really don't want to have to solder each time I want to update and love the "packaging" of this controller vs. a NodeMCU on a breadboard.  Am I doing something wrong or running into a hardware constraint?  In case it's useful here's free memory from the JSON API:  "freeheap":10600

Thank you for any assistance,
Brad

Def3nder, 2020-01-24:
Hi @bradmck, yes OTA does not seem to work on ESP8285 boards ,- I did not get it to work either.
On ESP8266 with 4M flash (like WeMos D1 mini) OTA does work perfectly.

According to the [ESP8266 documentation](https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#update-process-memory-view) there is a memory problem because the available memory must be at least as big as the new sketch is, but I tried to update with the [ESP8266MinimalHTTPUpdater](https://github.com/Aircoookie/ESP8266MinimalHTTPUpdater) what would definitively fit in the free flash and it did not work either.

@Aircoookie, do you have a suggestion how to get OTA to work with 1M ESPs ?

srg74, 2020-01-24:
You can try manual update described here https://bitbucket.org/xoseperez/espurna-original/wiki/OTA
Some reports saying it might help.

Def3nder, 2020-01-24:
@srg74, yes, this does work well with `ESPurna` and with `Tasmota` as well, but for some unknow reason this does not work with WLED when using `1M ESP8285 devices`.

srg74, 2020-01-24:
Oh, I have to check on my own. Maybe I made a mistake stating that is possible. As always, measure twice - cut once :) Sorry

Aircoookie, 2020-01-24:
Did you try the 1M_ota binary already? That one should have enought free memory for OTA to work on an ESP8285 :)

bradmck, 2020-01-24:
I just tried the binary you suggested....  flashed it wired first and then was able to do a manual OTA no problem.  Thank you for that - not sure why I didn't think to try the Release binary.  I've been trying to compile my own since I tend to update my others as you're updating code without waiting for releases.  Is it by chance documented anywhere how you compile that 1M_ota binary so I can do the same?  I'm super happy that I can use this device at all so no biggie if not.  

Thanks again for the work you put into this awesome project - I've recommended it to others several times.  My issue can be closed.  Hopefully someone finds the info I posted on flashing useful as it drove me crazy.

Def3nder, 2020-01-26:
@Hi @bradmck, you can close the issue youself - there is a button "close issue" at the left side of the comment button 😄

closed 2020-01-26 by bradmck