wled/WLED #622: Button wiring D1 Mini and NodeMCU

ggrant111, 2020-01-24:
First a big shout out to AirCookie and the other developers of this software. I used it for my christmas shows this past month integrating with xlights as well!!

I am trying to add a button to trigger a few macros and tried hooking it up as the first few paragraphs say in the Wiki adding a normally open button to D4 - GPIO0(i belive it says) and the other end to ground. When i did this and enabled the button in the settings and set the short, long and double press and the macros it didn't respond. I was able to change the macros the the HTTP rest api fine from a web browser. I tried searching in the issues to see if someone else had issues and found one that said it needed to be connected to D4 pin and VIN with a pullup resistor. So i am confused now because the D4 pin is supposed to already have a pull-up resistor. Where I am confused is the Wiki says one to the pin and one to ground and someone else said to the pin and the other end to VIN.

I am using D1 mini boards and also Nodemcu devkit1 boards as well.

Can someone help me figure this out?

thank you

srg74, 2020-01-24:
Hi! You have to use a pull-up resistor 10K from pin to 3.3V and then NO button from pin to ground. I'm using default button pin in my board and it's working just fine.
Best regards

ggrant111, 2020-01-24:
What do you mean default button? The on board button?

srg74, 2020-01-24:
Default button to WLED project e.g. GPIO0
Best regards

ggrant111, 2020-01-24:
Ok thank you

ggrant111, 2020-01-26:
Just soldered a resistor and button leads. After testing it works as
expected.

Thank you for the clarification.

srg74, 2020-01-26:
Hi! Np with button :)
Please close this issue if you satisfied. 
Best regards

closed 2020-01-26 by ggrant111