wled/WLED #627: Play tpm2 files with sd card

harueg, 2020-01-25:
Really great software !  Thank you !

Would it be possible to read tpm2 files from a sd card.

file format :  https://www.ledstyles.de/index.php?thread/18969-tpm2-protokoll-zur-matrix-lichtsteuerung/

Existing stand alone software  : https://github.com/TheChief79/tpm2arduino

https://hackaday.io/project/5714-glediator-from-sdcard-arduino

Thanks !

Aircoookie, 2020-01-26:
Hi, that indeed seems like a very good addition. Additionally I've received a request for being able to playback .fseq files (LED data format xLights uses), so those two feature requests are very similar except for the specific protocol used!

harueg, 2020-02-03:
Yep, that would be great  both protocols should work for me.
Thank you

harueg, 2020-06-01:
Hi @Aircoookie 
Any news on this topic ?

Thanks

Harald

Aircoookie, 2020-06-01:
Hi!
No, sorry. I know it's been a while and this is definitely coming, but I'll need to add some other features first, like basic file sysyem support.

harueg, 2020-06-01:
Ah, ok.
So I will keep looking forward to it !

Thanks and good night
Harald

Legsmaniac, 2020-06-01:
Cool!  I'll look forward to this.
This is something that fits nicely in with my recent requests too.
Thank you.

timdonovanuk, 2020-12-08:
+1 to this! Would be cool to see FSEQ format support in WLED - you could then just create an effect/show in XLights and upload and have it appear as an effect

pbolduc, 2020-12-19:
Related issue #1278

harueg, 2021-03-18:
Hi @Aircoookie 

Just trying, well a programmer, to get this fseqlib modified to neopixel(bus), because fastled still doesn't support rgbw.
That's trickier than thought.  Would be great, if that implementation in WLED would work some when.
Some pictures from a recent project with "glitter" effect.
[image]
[image]
[image]

processor83, 2021-03-19:
+ for this feature and if it happen would be super nice if sych feature would include this or FPP remote mode would be crazy :P.

harueg, 2021-03-21:
Hi @processor83  Hi @Aircoookie 

I think playing fseq would be sufficiently awesome.  
Integrating it as an effect would be the easiest from the UI side think, speed and brightness setting are already there.
Each fseq file on the sd card would be a separate effect.
How is FPP remote working ? How would you integrate WLED into Falcon Player ?

processor83, 2021-03-23:
There is the fpp remote protocol I guess when a device is setup as a fpp remote the master detect it automatically and it send information to the remote which sequence to play and some packer to keep it in synch.

It could be a fseq effect you select the. You have a drop down menu for options to choose a sequence and play it in loop or to shuffle or fpp remote and in fpp remote it would follow the fpp master. Falcon pi player is a sequence player and interface for some controller but i don't talk about a falcon controller itself.  Espixelstick have this fpp remote feature and I think they developed the library for this. I think wled ddp library used in wled come from espixelstick.

constant-flow, 2022-09-03:
Check [PR comment here](https://github.com/Aircoookie/WLED/pull/2413#issuecomment-1236206323) for most recent update on this 😀

It's now implemented as two usermods (SD & Playback)

blazoncek, 2022-11-25:
Thanks to @constant-flow SD card support has been added in #2877