wled/WLED #628: More than one lamp on a WLED AP

harueg, 2020-01-25:
Hi !

I would like to build a lamp with just mere dimmable white  and color effects.
So the white would need it's own channel.
With a Meanwell LED driver that would just be any pin, which could give a ~ 2kHz PWM output.

Is there any planning of a multichannel WLED ?
Now I need one WLED ESP32  per lamp, or did I get something wrong ?

Thanks

tonyn0, 2020-01-26:
If I have you right, they call this "analog" LED support and it's in there. Support for RGBW[W] support is also in there.

Def3nder, 2020-01-26:
Hi @harueg, @tonyn0 is right: this is the "analog LED support" and it can be enabled either using `#define WLED_USE_ANALOG_LEDS` (have a look in `NpbWrapper.h`) or - if you use PlatformIO simple compile the device `esp8285_4CH_MagicHome`, `esp8285_4CH_H801` or `esp8285_5CH_H801`.
These three have fixed PINs for the LED channels.
The MagicHome LED controller is prepared for the PINs:
```
#define RPIN 5   //R pin for analog LED strip
#define GPIN 12   //G pin for analog LED strip
#define BPIN 15   //B pin for analog LED strip
#define WPIN 13   //W pin for analog LED strip (W1: 14, W2: 04)
```

And the H801 Wifi uses:
```
#define RPIN 15   //R pin for analog LED strip
#define GPIN 13   //G pin for analog LED strip
#define BPIN 12   //B pin for analog LED strip
#define WPIN 14   //W pin for analog LED strip (W1: 14, W2: 04)
#define W2PIN 04  //W2 pin for analog LED strip
```

The PWM frequency used for an ESP8266 is set to 880 Hz (see line 179 in `NpbWrapper.h`: `analogWriteFreq(880);` ) and for an ESP32 it's set to 5kHz (see lines 151++: `ledcSetup(0, 5000, 8);`)

Right now you need one ESP per LED strip.
An ESP32 has 16 channels for the LEDC driver so it would be possible to drive three RGBWCT LED strips...
...but this is not implemented yet.

harueg, 2020-01-26:
Yes, i found it. It's in the b2 release.
Unfortunately I get a compiler error :
NeoPixelWrapper::_pGrbw' will be initialized after [-Werror=reorder]
Line 296 in NpbWrapper.h
using Arduino IDE

Will these settings appear on the app as well ?
In my application I will have WS2812B RGB Strip and a separate white LED strip on  a standard CV LED driver ( some meanwell ). These have a PWM signal input 100Hz to 3kHz.

On the WS2812B RGB Strip there should be some animation running, fed by a tpm2 file.
see Issue #627

As boards I either have TTGO-T Display  or heltec wifi kit32, will it work on them ?
when the compiling issue is solved.

Due to the low price of the ESp32s, there is no problem using 2 controller, it's more a matter of comfort.

Thanks a lot !

Aircoookie, 2020-01-28:
@harueg have you installed all the libraries from the Quick start guide? What is your NeoPixelBus version?

Yes, sure, WLED should run on any ESP8266 or ESP32 based board :)

harueg, 2020-01-28:
@Aircoookie   for the esp32 there is only one mentioned : Arduino Core v1.0.4
the neopixelbus is v2.5.6  Arduino IDE 1.8.10

Thank you

Def3nder, 2020-01-28:
Hi @harueg, another quite easy way is to install **VSCode** and then the **PlatformIO extension**.

Having done this you only need to open the WLED main folder where you cloned the repo in and hit the ✔️ in the status line at the bottom:
[image]

The good thing with this is that PlatformIO does handle all the library constrains on a "per-project" basis (while the Arduino IDE does this globally for all projects).

harueg, 2020-01-28:
Hi @Def3nder 

for me it stops there with being " quite easy ".
Had a few tries with vscode / platformio ...  too many options.
Settings are in the platformio.ini ?

Do I have to have all libraries and environments in the wled-master folder ?
as described in the platformio.ini ?

Thanks

harueg, 2020-01-28:
[platformio.txt](https://github.com/Aircoookie/WLED/files/4123842/platformio.txt)

Def3nder, 2020-01-29:
Hi @harueg, no - even more "easy". You do not have to care about libraries at all.
Your modification of the platformio.ini is correct (only the default environment changed to `default_envs = esp32dev` ). Then you need to tell the compiler that you want to use the analog LEDs in the selected environment. 
So scroll down to `[env:esp32dev]` and add this line below `build_flags =`:
`-D WLED_USE_ANALOG_LEDS`

[platformio.txt](https://github.com/Aircoookie/WLED/files/4127062/platformio.txt)

With these changes you get a binary for the ESP32 with the following PINs activated:
```
#define LEDPIN 2  //strip pin. Any for ESP32
#define BTNPIN 0  //button pin. Needs to have pullup (gpio0 recommended)
#define IR_PIN 4  //infrared pin  
#define RPIN  5   //R pin for analog LED strip   
#define GPIN 12   //G pin for analog LED strip
#define BPIN 15   //B pin for analog LED strip
#define WPIN 13   //W pin for analog LED strip (needs to be enabled in the settings later)
```

Coming back to PlatformIO:
if you start VSCode and open the main folder "WLED" where you cloned the repo to (or if downloaded as ZIP, where you extracted the ZIP in), then all you need to do is pressing that ✔️ in the bottom left status line.

Does this work for you ?

harueg, 2020-01-29:
Hi @Def3nder 

Well, yeah, kind of working, managed to produce binaries with various settings.
You can choose the environment in Project Tasks, that worked
The analog RGB setting doesn't show up in LED settimgs on the UI, correct ?

Obviously I can have parallel support of WS2801 and anlog RGB ?
At least it did build it.

harueg, 2020-01-29:
yep, that works.

Now " just " trick the code, that even when set to RGBW in the UI  the output is RGB for PIN 2.
and  W for PIN 13.

Would that be possible, would it make sense at all ?
or better adding the slider and the code for a separate white slider ( or more sliders for more lights )

Def3nder, 2020-01-30:
@harueg: to your questions before: yes: analog LEDs is right now not in the UI settings (I [plan](https://github.com/Def3nder/WLED/projects/1) to add this) and: yes, analog and digital (WS2812 etc) strips can be used in parallel.

But: using RGB for the digital one and the white channel for the analog would not make sense, because the analog strip is "grabbed" only from the very first pixel, while the digital strip has differnet colors for each pixel. To use RGBW you need to get an SK6812 strip that supports 4 color (like [this](https://de.aliexpress.com/item/32476317187.html?spm=a2g0o.productlist.0.0.6ee447d2P7GXbh)).

harueg, 2020-01-31:
Hello @Def3nder 

Yes, it takes the color values from the 1st LED. So that idea only works with a RGB Led strip and one white strip. 
the plan would be to set it on RGBW on the UI.  There we get the white slider and values for white.
But the board is just sending RGB to the LED strip on pin 2 . On pin 13 it is sending the white.
I guess it will be tricky in the code to calculate RGBW, but just send RGB.  ??

Def3nder, 2020-01-31:
Hi @harueg, from a coding perspective this would be easy. 
So, if you would like to have a binary that does this, I can build one for you.
(I would initialize the strip as RGB only and then grab the white channel in the SetPixelColor() function and send it to PIN 13. That should work).

So, you could then have the one strip for"RGB effects" and the other one for " white light only".

harueg, 2020-01-31:
Hi @Def3nder 
That would be great. I would compile the binary myself, since I am not sure about the white pin yet.
It is already in use on the TTGO T1 Display.

Thank you very much

harueg, 2020-02-17:
Hi @Def3nder 
Did you change / do anything ?
Tried to compile the code here, but having trouble with ESPAsyncTCP.
I did use the AsyncTCP, but that tells me, it is not made for Win64... strange.
https://github.com/me-no-dev/ESPAsyncWebServer/issues/505

Maybe you can make the binary and I get a normal ESP32.

Thanx and good night

Harald

Def3nder, 2020-02-18:
Hi @harueg,

had some problems with the build process, too, this morning - anyhow - deleting the .pio folder did solve this for me (I'm using VScode with PlatformIO extension).

I've created the ESP32 binary with these options:
```
no changes to LED type, so this will work with WS2812, SK6812 etc.
LED-PIN 2
IR-PIN 4
PIN for analog LED: 13
```
You need to enable the checkbox `LEDs are 4-channel type (RGBW)` in **LED settings**.
Then the slider for the brightness should show in the UI.
This slider should only controll the white channel attached to `PIN 13 for analog LEDs`.
**The digital strip has to be `RGB-only` (not RGBW).**

I could not test this as I'm at work right now - I wil try it this evening...

I hope you can flash this (I did not test flashing an ESP32 other than directly from PlatformIO or the Arduino IDE) - inside the ZIP there is a file firmware.bin - I think it should work to use this for a manual update (`http://[WLED-ip]/update`)
[WLED ESp32 pre-0.91 with analog W channel on PIN13.zip](https://github.com/Aircoookie/WLED/files/4218415/WLED.ESp32.pre-0.91.with.analog.W.channel.on.PIN13.zip)

harueg, 2020-02-18:
Hi @Def3nder 

Thank you, yeah work, stupid topic.

I was able to flash it, but there was no reaction on pin13.
btw.  pin13 is source ?  at least it worked the last time this way.
I must admit i didn't connect anything on pin2.

Segments :
But I still have to choose rgbw for the rgb strip, so I get 32bit Information for a 24bit LED.
that will shift the colors, 

Good night

Harald

Def3nder, 2020-02-19:
Hi @harueg,

yes, the "normal" behavior would be to get 32 bit colors, but I changed the code that way that it stays with 24 bit colors for the digital LEDs and only uses the additional 8 bits for the white channel on GPIO13.

Is this the board you use ?
[image]

harueg, 2020-02-19:
ah, ok

Then there is no need for segments?

Also tried this board :
[WIFI Kit 32 (1).pdf](https://github.com/Aircoookie/WLED/files/4226269/WIFI.Kit.32.1.pdf)

Def3nder, 2020-02-20:
Hi @harueg, yes - the slider will go directly to the white LED strip. And all the rest to the RGB strip.

I just saw that `GPIO4` is used for the TFT and it's defined for the `IR-PIN` as well.
This could be a source why this is not working.

Please try the following binary - I've disabled the `PIN 4` (and the IR support) in this release:
[WLED ESP32 pre-0.91 analog W channel on PIN13 no IR.zip](https://github.com/Aircoookie/WLED/files/4230183/WLED.ESP32.pre-0.91.analog.W.channel.on.PIN13.no.IR.zip)

harueg, 2020-02-20:
Hi @Def3nder 

Nope, doesn't work on either board.
With the heltec board I checked the output on pin 13 with another program, it lightens, dims and fades.
Other thing with WLED AP, even when I set a IP address for it, it decides to use any IP address.
Do I have to disable DHCP separately ?

Completely other topic, just slightly related.
I want to build an enlarged copy of these LED candles.
https://techtest.org/die-besten-kerzen-mit-beweglicher-flamme-die-air-zuker-led-kerzen-im-test-frohe-weihnachten/
The candle flame moves with a magnetic field and various LEDs are changing their brightness ( not color )
This by itself is a kind of easy, except the maybe necessary parallel ( multi )tasking aspect.
The attached sketch a kind of worked, though I am not sure whether freertos works properly on such a little computer.
Anyway, this should run on an ESP32  with some kind of remote control.
ON / OFF, overall brightness, frequency and amplitude of the effects.

Would it be possible to integrate it into WLED as an effect ?
It would need 7 pins as output 3 for the magnetic coils and 4 for the LEDs.

How much would that be in time and money ?
The resulting code should still be open source.

I suppose you heard of the espressif mesh
https://www.espressif.com/sites/default/files/documentation/esp32-meshkit-light_user_guide_en.pdf
This is different from the MQTT with WLED ?

[blink_analogread_Led_ino5.zip](https://github.com/Aircoookie/WLED/files/4232601/blink_analogread_Led_ino5.zip)

other promising tries :
[candle_millis_2.zip](https://github.com/Aircoookie/WLED/files/4232759/candle_millis_2.zip)
[candle_millis_1_random.zip](https://github.com/Aircoookie/WLED/files/4232762/candle_millis_1_random.zip)
 

Thank you

Harald

harueg, 2020-02-20:
btw.  I ordered some normal ESP32s, just to check.

harueg, 2020-02-20:
I checked the other analogue outputs, there is just a very dim glim.
The normal 2812b LEDs are working perfectly.

Good Night

Harald

harueg, 2020-02-21:
Hi
For some reason reason I managed to compile it again.
And ...   It's working  I even managed to get the oled display working, so I can see the actual IP address.
I also used another board : 
https://www.az-delivery.de/products/esp-32-dev-kit-c-v4?_pos=1&_sid=7c21c9af4&_ss=r

Did you uncomment line 10 in NpbWrapper.h ?
I guess some of the effects override the white settings.  White won't work with pride2015.

Good Night

Harald

Def3nder, 2020-02-28:
Hi @harueg, no I did nothing with line 10 - this would switch from WS281x type LEDs to the old WS2801. 

And: yes, the effects that use FastLED palettes don't work with RGBW - they use RGB only.

harueg, 2020-02-29:
Hi @Def3nder 

Line 10 in NpbWrapper.h does matter, if commented analogue LEDs don't work.

10 //#define WLED_USE_ANALOG_LEDS //Uncomment for using "dumb" PWM controlled LEDs (see pins below, default R: gpio5, G: 12, B: 15, W: 13)

harueg, 2020-02-29:
So WLED uses NeoPixelBus and FastLed ? 
These libraries or programs won't be useful for playing pixelvideo and having white light ?
https://github.com/ShaunPrice/FSEQLib
https://github.com/TheChief79/tpm2arduino
https://hackaday.io/project/5714-glediator-from-sdcard-arduino

Thanks and good night

Harald

nelzon101, 2020-06-03:
> Hallo @harueg , aus Sicht der Codierung wäre dies einfach.
> Wenn Sie also eine Binärdatei haben möchten, die dies tut, kann ich eine für Sie erstellen.
> (Ich würde den Streifen nur als RGB initialisieren und dann den weißen Kanal in der SetPixelColor () -Funktion abrufen und an PIN 13 senden. Das sollte funktionieren.)
> 
> Sie könnten also den einen Streifen für "RGB-Effekte" und den anderen für "nur weißes Licht" verwenden.

Hello Def3nder , I love this project! What did you change where in the code? I want to transfer the settings for a Wemos D1 mini. One pin for digital RGB and one pin for analoge white. Thanks

harueg, 2020-06-03:
Hallo @nelzon101 

Vielen Dank das habe ich bereits kompiliert und hat auch funktioniert.
Die nächste Hürde zur fertigen Lampe wäre jetzt das Abspielen von fseq Dateien die dann aber nur den RGB Ausgang benutzen. siehe issue #627 

Noch einen schönen Abend

Harald

nelzon101, 2020-06-04:
Hallo Harald. Ich würde gerne selbst kompilieren (VSCode, aktuelle Wled-Version) und folgendes umsetzen wollen:  "(Ich würde den Streifen nur als RGB initialisieren und dann den weißen Kanal in der SetPixelColor () -Funktion abrufen und an PIN 13 senden. Das sollte funktionieren.)" 

Wo passe ich das im Code an?

Gruß Nelzon

harueg, 2020-06-04:
Hallo @nelzon101 

Das ist jetzt schon wieder so lange her.
Aber eigentlich ist alles weiter oben im Text beschrieben. An der SetPixelColor () habe ich nichts geändert. Ob das so bei dem ESP8266 funktioniert weiß ich nicht. 

Beste Grüße

Harald

harueg, 2020-06-04:
Schau mal hier 
https://wled.discourse.group/t/analog-led-support/55

nelzon101, 2020-06-04:
Danke, aber das ist nicht das selbe. 
Folgendes Ziel habe ich (Basis ESP2866): RGB **nur** für WS2811 (Farben und Effekte) & White mit einem Slider für den analogen Whitestrip

theoretisch also RGBW wobei RGB an WS2811 & W an PWM geschickt wird.

@Def3nder please let see your changes in your .bin file below ;)

closed 2020-10-10 by stale[bot]