wled/WLED #629: No WiFi with GPIO2 (NodeMCU/Wemos pin D4) connected

nmsoares, 2020-01-26:
I'm running WLED_0.9.0-b1_ESP8266.bin on a Wemo D1 Mini with a static IP configured. I cannot get wifi/network connection if the GPIO2 is connected to the board. If I reboot with the GPIO2 upluged thenI get a network connection.

Def3nder, 2020-01-26:
Hi @nmsoares, did you flash with "clear flash" switch ?
There is a similar issue #592 - one solution was to flash the release 0.8.6 and then the 0.9.0b1 with manual OTA over the settings-page.

Does this work for you ?

Aircoookie, 2020-01-26:
In addition to what @Def3nder suggested, do the LEDs light up as they should (orange by default)? If not, your LEDs are likely pulling gpio2 low and preventing the d1 from booting. You could try adding a 4.7k (or similar value) resistor between gpio2 and the 3.3v pin :)

L337Lobster, 2020-04-13:
Can confirm, 4k7 resistor between D4(GPIO2) and 3.3v allowed my nodeMCU to boot properly even with my LED data wire connected to D4(GPIO2). This tells me that, at least in my case, GPIO2 was being pulled low! Hope confirmation of this fix, at least for me, helps someone else!

L337Lobster, 2020-04-25:
Update, when I add a logic shifter the problem comes back even with the resistor between GPIO2 and 3.3v, any suggestions?

Aircoookie, 2020-04-25:
@L337Lobster from the images and description this particular shifter might have pull-downs on the inputs that counteract the pull-up resistor. You will probably either have to use a different pin (like gpio3 or 4) or a different shifter (my favorite is the SN74AHCT125, though be careful, it needs 5v supply voltage and also only outputs 5v - which should still be plenty for any 12v LED though)

closed 2020-08-30 by stale[bot]

jordonezb-milenium, 2023-01-03:
I have the same problem with an esp8266, I don't understand adding a resistor between pin 3v3 and D4, a drawing of the schematic would be of great help thks

blazoncek, 2023-01-04:
Just connect 4k7 Ohmm resistor, one wire to GPIO2 and another to +3.3V power.

jordonezb-milenium, 2023-01-06:
I do not understand very well and I tried to place the resistance but it does not work correctly, I'm very new