wled/WLED #630: Transitions not working with home assistant

vmazmaz, 2020-01-26:
Hello,

First of all, I would like to thank you Air cookie for the phenomenal job you have been doing with WLED, it has been a game changer for DIY LED lighting. 
I have been trying to use transition times with Home Assistant (using the native integration) but I have not been successful, the state changes immediately, regardless of the `transition:` value. Am I missing something ?

Thank you !

fishbone-git, 2020-01-26:
see #543

vmazmaz, 2020-01-26:
In my case, the `transition:` parameter is ineffective, even inside of an automation.

Aircoookie, 2020-01-26:
Hi and thank you for your kind words!

What home assistant version are you running? This should have been fixed in version 0.104.0 :)

vmazmaz, 2020-01-28:
I upgraded to 0.140.0 and it is working now ! I still experience a brief flash a the beginning of the transition but I suspect it might not have anything to do with WLED but rather with the Circadian Lighting component I am using. Thanks !

Aircoookie, 2020-01-28:
@vmazmaz great that it is working now! Does the flash also happen when you turn on the light with the web UI? If yes, it is most likely an issue with the signal level not being strong enough to reliably drive the LEDs. If that is the case, you could try adding a level shifter.

closed 2020-01-28 by Aircoookie