wled/WLED #631: Question: switch power to LEDstrip using WLED controller and relay: how?

raoulteeuwen, 2020-01-26:
I read a couple of tickets about using a relay, like https://github.com/Aircoookie/WLED/issues/301 . 

I currently have 2 strips, one with a NodeMCU , one with a Wemos D1 which both run WLED. I'm also using Home Assistant.

I read LED strips still consume power when 'off', so for now, i use a Sonoff Basic (on one strip) and a Tuya-kind of smart plug to totally switch off power to the strip and NodeMCU/Wemos. So my automation from HA basically is: switch on the Sonoff/Tuya, wait a couple of seconds, send any commands to the WLED NodeMCU/Wemos.

This works, but means for every LED strip i need 2 devices. Would it make more sense to connect a relay to the NodeMCU/Wemos, and use the relay to switch on/off power to the LED-strip? What are advantages/disadvantages?

If it is recommended instead of my current config, can someone help me a little? Never used a relay: am i right i need to to run 5v and gnd to the relay, and connecting a pin from the NodeMCU/Wemos (D6) to the relay to switch/control the relay? And connect the LED-strip to the other side of the relay, running either gnd or live through the relay?

How does the software side work? Do the standard bin-files have this option, or do you need to compile your own version to control the relay? How do you control the relay (MQTT-commands, GUI...)?

Aircoookie, 2020-01-26:
Don't worry, using a relay in WLED is now fully supported and should offer a great replacement for your current setup, which isn't bad, but some simplification would indeed be nice!

Yes, that's true, they use about a milliamp per LED when unlit, which can add up quickly when running more than 100 or so LEDs. Setting it up is quite easy, just get a generic Arduino relay shield like [this one](https://www.ebay.com/itm/One-1-Channel-Relay-Module-Shield-5V-Control-250V-10A-w-Optocoupler-for-Arduino/382572428968) and connect its input to GPIO12 (D6).
Advantages are definitely the reduced power consumption (+what you save by not running a second ESP or smart plug), disadvantages are that it only works up to ten amps (the relay rating).

You can choose whether you want the relay to interrupt the LED 5V power line or the live wire to the LED power supply. I would use the former for safety reasons and since you don't need another power supply for the ESP.

Yes, D6 as relay pin is enabled by default in all recent binaries :) The relay operates automatically (turns on when LEDs are on and off a second after lights are fully off), but that is basically all you need, so there is no manual control of the relay possible at this point, it always depends on the WLED power state.

Hope I was able to help and let me know how it goes or if you have more questions!

raoulteeuwen, 2020-01-27:
Will try it somewhere in the future/when i find the time. Thank you @Aircoookie !

closed 2020-01-27 by raoulteeuwen

Dreamoffice, 2020-03-01:
@Aircoookie  is it possible to swap the logic of GPIO12. my relay needs to get ground to power on. actually the D1 mini is serving 3.3 volt on GPIO12 if the LEDs are switched on

austwhite, 2020-03-01:
The relay can be inverted in the script if you want to compile yourself in the idea.
Either that or get a relay that can be inverted.  Some have a jumper that can switch the logic.

Dreamoffice, 2020-03-02:
@austwhite where can i change it? i tried it but no success

raoulteeuwen, 2020-03-26:
(sorry for the long post)
Just checking. I have, for me for the 1st time, tried to use a relay. I'm using a Wemos D1 mini, which has only 1 GND pin. Afaik, i need gnd to run from the PSU to a) the LEDstrip, b) the Wemos, c) the relay (control side). I have now wired up as can be seen in https://photos.app.goo.gl/waCYwMe6zSRF6gyZ6 
[image]
When i connect the plug to a power source, the Wemos boots, WLED starts, the red LED on the relay turns on (green is off) and the LED-strip behaves as expected (doing the sequence i configured in WLED). When i switch off WLED via a browser (pressing on/off button), the LED-strip turns off, and on the relay the green LED turns on (red stays on as well).

On the relay, one wire (from the power supply plug) goes to COMM, and i have a wire running from NC to the LED-strip. 

Although everything seems to work, i'm just a little unsure whether i accomplish what my intention with adding the relay is: making sure the LEDstrip doesn't use power when WLED is off. I'm confused by the green LED turning on. I've tried to measure power running over the system with a multi meter, but i'm such a noob with this, that when i tried, i seem to short something, the Wemos turns off, and when i remove a probe, everything turns on again. I've resorted to a smart plug, and i do see low electricity use when the strip is off, just not sure whether it can be even lower (the Wemos keeps being powered off course, and maybe there is a little loss in the relay). I measure about 1W (and 224 V and .02 A). When i, while the LED-strip is on, unplug the wire that enables the Wemos to control the relay, and than switch off WLED, the LEDstrip switches off, the relay does not change state (nor click), and i measure 1.6W (and 224V and .02A).

One more thing i tried to verify correct working: while everything was off (but powered, so the relay should make sure there is no power to the LED strip, the green led on the relay was on), i unplugged the wire from the relay that the Wemos/WLED uses to control the relay. I hoped the relay would retain its state, but when i unplug the control wire, i hear it click, and the green LED turns off. 

Can anyone confirm this looks o.k. / seems o.k. / relay probably works as intended :-)?

raoulteeuwen, 2020-03-28:
@Aircoookie one question please: i have a circuit with a relay. When i press the Power-button in WLED, i hear the relay click, the green LED on the relay switches on, the LED strip turns off. Am i right that current D6 behaviour in WLED is to turn on D6 (am i right the correct language is: switch the pin to high?) when WLED is switched off? (when i use the COMM and NO on the relay, the LED strip never lights, so i use COMM and NC)

Aircoookie, 2020-03-28:
@raoulteeuwen I just checked in the code and the default behavior is to turn D6 on (high) when the LEDs are on and off (low) when the LEDs are off. It can be changed though with the `RLYMDE` define in `NbpWrapper.h`. Many relays are powered when the signal is LOW, so that is probably the case for you.

raoulteeuwen, 2020-03-29:
Tried to compile from master (haven't done that a lot, but was able to compile without errors, so seemed ok). Change RLYMDE from 1 to 0 in NbpWrapper.h. After uploading, the Wemos started to flip flop its blue LED every 1.5 sec or so, i heard a relay kind of click every time, but this made the whole config unaccessible. So went back to the compiled bin and leave it like this for now (is working fine, just now using the NC port on the relay, which means WemosD1 with WLED is powering the relay when WLED is off: would rather had it the other way around ... will look into another relay as well)

Aircoookie, 2020-03-29:
Just tried it and RLYMDE 0 seems to work fine for me, rather I believe that you are having bootloops with the version you compiled. Here is a rlymde 0 binary, you can try if that works :)
https://drive.google.com/open?id=1TgOx3Va1OA2GBNZmgMgoSj0cJufyjEeq

raoulteeuwen, 2020-03-29:
Guess you were right, and that bin works, thank you, i can now use the NO port, and the relay only gets powered when WLED is on. Buying you a <beer/coffee/tea...>. Will search what could cause that boot loop (bad flash, wrong baud rate...)

Aircoookie, 2020-03-29:
Awesome, thank you for the coffee :)
Actually the most common cause for bootloops is Lwip 2, which ist the default in Arduino IDE. It works best if you use these settings: https://github.com/Aircoookie/WLED/wiki#compile-settings
If you are already using them, maybe a library is outdated or the flash layout is wrong...

Warsenius, 2020-07-19:
reversing the relay pin might be a nice feature to have a checkbox for in the webinterface.
since triggering the relay also takes a bit of power if depends if your relay is on or off most of the day what is most efficient per case.

Chaosflori25, 2020-09-13:
Could you reupload a binary for 10.2 again? @Aircoookie I tried to compile it but it did not work and after 2,5h I can't find anything that I can change for it to work. Thanks a lot in advance.

Aircoookie, 2020-09-13:
@Chaosflori25 here you go (assuming you were asking for a ESP8266 binary with RLYMDE 0)
https://drive.google.com/file/d/1BuXCWeDjRP9R8om7s3zuceRMQfFvbhug/view?usp=sharing

fpovoski, 2020-09-28:
I was wondering if there is a built-in delay from the time you fire GPIO12 for the relay until you actually start pushing data to the LEDS. Reason is this; when using high led counts and a large supply you actually want to switch on that supply (separate small supply for ESP of course) at mains which takes some time to ramp to nominal voltage. This eliminates the "idle" draw of about 15 watts of a large supply being on continuously, plus of course "idle" current of the LEDS. Otherwise you see strange behavior on the undersupplied string until supply ramps.  By the way, **very nice software** but in my initial testing last year on my current hardware which is very reliable using my custom software I was getting some undesirable "flashing" effects with WLED.  But I shortly will be taking one unit offline and will give WLED another test.  Thanks and I'll keep you updates

Aircoookie, 2020-09-28:
@fpovoski no, there isn't yet but I agree that at least like 50ms are needed for the power to stabilize before sending data. There is an FR for this: https://github.com/Aircoookie/WLED/issues/346

For getting rid of the flashing, you'll most likely need a level shifter to convert the 3.3v data signal to 5v. If your own software doesn't exhibit this flashing, it is probably either running the LEDs at a higher brightness (causing voltage drop and the voltage differential between LED supply voltage and data signal voltage to lower, thus making the signal readable to the LEDs) or at a lower framerate than the 42 fps WLED uses for most effects.
https://github.com/Aircoookie/WLED/wiki/FAQ#reason-2

fpovoski, 2020-09-28:
I do level shifting the using the proper level shifting 74AHCT125/series resister and very short run to first LED.  I run 1000 LEDs per controller using a tweaked FastLED (to optimize for any delay) and achieve a high frame rate with APA102s.  I have very robust power distribution along the strips using 10 gauge wire, 5V/60 amp supply for each.  I’ll give your sw a flash and see if I see that problem again.  Also with the turn on delay a programable time might be nice since different supplies ramp differently. But I will give it a measure under a slight load.  This was the one I used in past:
https://www.amazon.com/gp/product/B017YEOAPA/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

and a current one

https://www.amazon.com/SHNITPWR-Adapter-Converter-Transformer-Lighting/dp/B07TXWFSZ9/ref=psdc_10967761_t1_B017YEOAPA
Thanks

https://github.com/Aircoookie/WLED/wiki/FAQ#reason-2

Chaosflori25, 2021-02-02:
Hello all,
I again tried to compile the sketch but it failed again. Could somebody compile the latest Update for WLED with RLYMDE 0 for the esp8266? Thanks a lot. I even reinstalled everything but I think there is something I'm doing wrong.

mortenx, 2021-07-02:
Where can i see relay button in wled app?

Aircoookie, 2021-07-02:
@mortenx you can set the Relay pin and mode in LED settings/Hardware Setup if you are running WLED 0.12.0 or newer.

snorrebot, 2021-07-09:
Hi all, i'm also struggling with the 5v relay. Would be wonderful if someone could help me out...

- relay is properly connected with pin d6 (see attachment)
- in the settings menu I entered the right pin for the relay
- I press ON in the app and the relay clicks nicely to on.
**- I press OFF and the relay stays on**, nothing changes.

The D6 pin just hangs with the ON signal, even after a restart of my ESP. Only when I change the relay pin in the settings menu to a different pin, I can start over and turn ON the relay, but again I can't turn it off. It feels like a software problem but I can't figure it out....

Thanks for your help!

[image]

[image]

raoulteeuwen, 2021-07-09:
> 
> Hi all, i'm also struggling with the 5v relay. Would be wonderful if someone could help me out...
> 

Just checking: if ON, does WLED still work, and does it change your LED-strip? Did you try to change the wires on the relay to check whether NC/NO is making any difference (only needs 1 wire changed on the output side)? Does wiring the relay with ground directly from the power supply make any difference?

snorrebot, 2021-07-09:
Thanks for your quick response! 
- if ON WLED works as it should, ledstrip works as intented
- I changed between NC and NO, NO did not work
- all the ground cables are connected in one chain together to the power supply

snorrebot, 2021-07-14:
No one? It feels like a software bug because the D6 pin for the relay just keeps the ON signal forever . If I remove the cable from pin D6 my relay immediately switches off as it should...

RichieRich69, 2021-07-30:
Hi @snorrebot, I was a similar issue a while ago. Turns out hyperion/led was keeping WLED on despite it not showing streaming on the UI. Your wiring looks okay. Also test the relay separately, I had a defective one too. Finally the latest version has a pull high / low in the config it's worth trying both.

ChuckMash, 2021-08-29:
@snorrebot I ran into this exact issue, with this exact (or extremely similar) relay module.
I was able to solve the issue by adding a 4.7K ohm resistor between D6 and the relay input.

I'm not terribly sure why this was necessary for this relay module, so I ended up selecting a different "higher quality" relay module that didn't have this problem.

blazoncek, 2021-08-29:
The relay module in question is a 5V relay, taking 5V logic as input. This is important since ESP can only provide 3.3V.
This module also features negative logic whic means that relay is switched off when input pin is HIGH (5V).

What you need to do is add simple MOSFET between D6 and relay input. Don't forget to add 10k pull-down resistor to D6 and 4k7 resistor between 5V and drain on MOSFET, connect source to GND.
I can recommend 2N7000 as a good choice for MOSFET.

snorrebot, 2021-08-29:
Thanks ChuckMash and blazoncek for the great options! 
@ChuckMash do you have a model number or link for your working relay module?

blazoncek, 2021-08-29:
Another option is to use level shifter in the same way as for LEDs.

ChuckMash, 2021-08-29:
@snorrebot The 5v Relay module by HiLetgo.

supss, 2021-09-14:
hello all, great to see relays being supported, but is there a way to also have an external switch (like a momentary switch).  logically this would be considered a 3 way setup so that the LEDs can be toggled by both the UI and the physical switch?

blazoncek, 2021-09-14:
> hello all, great to see relays being supported, but is there a way to also have an external switch (like a momentary switch). logically this would be considered a 3 way setup so that the LEDs can be toggled by both the UI and the physical switch?

These are supported as well.

vxav, 2023-03-05:
> Don't worry, using a relay in WLED is now fully supported and should offer a great replacement for your current setup, which isn't bad, but some simplification would indeed be nice!
>
> Yes, that's [...]

Hey @Aircoookie, you mention that it’s safer to interrupt  the led power line as opposed to the psu live wire. Why is that? I would have assumed the opposite.

glebmiller, 2024-01-31:
> Hi all, i'm also struggling with the 5v relay. Would be wonderful if someone could help me out...
>
> * relay is properly connected with pin d6 (see attachment)
> * in the settings menu I entered the [...]

I had the same problem, solved it by powering relay from 3.3v, seems that the relay only works when both IN and VCC has same voltage. (GPIO from esp or nodemcu gives 3.3v)