wled/WLED #632: WLED Aircoookie mit Odroid N2 möglich?

Cosmicbase, 2020-01-26:
Hallo, ich habe einen Odroid N2 mit aktuellen CoreElec, Arduino Uno R3, APA102 am laufen. 
Kann ich durch Tausch des Arduino in einen ESP32 meine LEDs ins Netzwerk einbinden?
Preislich wäre der Austausch mehr als rentabel. 
Wie schaut es mit dem Hyperion.ng aus wo ich momentan nutze?

Aircoookie, 2020-01-26:
Hi und willkommen!
Wie sieht denn das Setup aus? Wenn der Arduino Uno nur die Daten von Hyperion (ich nehme an dies läuft auf dem Odroid) via USB entgegenimmt, kann der WLED ESP32 dies auch (über das UDP Typ 0 Protokoll, also kabellos).

Def3nder, 2020-01-27:
Hi @Cosmicbase, hyperion.ng spricht den Arduino als Adalight über USB an, oder ?
Wenn ja, dann kannst Du den Arduino durch einen ESP8266 oder einen ESP32 austauschen 😄

Cosmicbase, 2020-01-27:
> Hi @Cosmicbase, hyperion.ng spricht den Arduino als Adalight über USB an, oder ?
> Wenn ja, dann kannst Du den Arduino durch einen ESP8266 oder einen ESP32 austauschen

OK danke, Ja mein Arduino ist via USB am Odroid N2 und wir als Adalight angesprochen. 

Gibt es eine klare Empfehlung in Sachen Stabilität und momentanen Standpunkt?
ESP32 oder ESP8266?

Cosmicbase, 2020-01-27:
> Hi und willkommen!
> Wie sieht denn das Setup aus? Wenn der Arduino Uno nur die Daten von Hyperion (ich nehme an dies läuft auf dem Odroid) via USB entgegenimmt, kann der WLED ESP32 dies auch (über das UDP Typ 0 Protokoll, also kabellos).

Ja soweit hast du recht. Ich gehe davon aus das Ausgabe in dem Scroll Down "Steuerungstyp" ausgewählt wird siehe Bild
[image]

closed 2020-06-02 by stale[bot]