wled/WLED #633: Stuck Loading WLED UI...

sT0n3r, 2020-01-27:
Hi

I have a small problem, this is my first install of this amzing software and i love it so much thank you for all your hard work. i am using a wemos d1 mini to control some 3d printed nanoleaf like panels with ws2812b leds , wled is perfect for what i need, i installed the latest version WLED_0.9.0-b1_ESP8266 using NodeMCU-PyFlasher-4.0-x64 everything is working great except it could not auto detect the light and on the wled app when i goto the wled control panel it just says Loading WLED UI... , with all the options ghosted underneeth, i can click the ghosted "config" ( i cant click power/timer/sync/preview) and goto the config screen and see and edit the options no problemi, i even got to test the ambilight function using prismatik which is working perfectley, i can access the device using mdns from my pc on firefox and can see and use all the functions correctley, my phone is a bit old its a elephone p8000 running android 6, i just thought i would let you know, keep up the fantastic work

Def3nder, 2020-01-27:
Hi @sT0n3r, sounds wired 🤕.
Someone did report that the wifi connection to an ESP8266 is way better when switching to channel 11.

What does WLED display if you use firefox and enter `http://[WLED-IP]/json/info` as value for **"signal"** ?

sT0n3r, 2020-01-28:
my /json/info looks like this

ver | "0.9.0-b1"
-- | --
vid | 1912121
leds |  
count | 6
rgbw | false
wv | false
pin |  
0 | 2
pwr | 161
maxpwr | 850
maxseg | 10
seglock | false
str | true
name | "WLED"
udpport | 21324
live | false
fxcount | 89
palcount | 51
wifi |  
bssid | "D8:D7:75:BB:8F:0A"
signal | 96
channel | 11
arch | "esp8266"
core | "2_6_2"
lwip | 1
freeheap | 15456
uptime | 194
opt | 119
brand | "WLED"
product | "DIY light"
btype | "bin"
mac | "b4e62d238e5f"

Aircoookie, 2020-01-28:
Hi, if it works in the browser, but not in the app, I believe it could be an issue with how your phone handles the WebView implementation (normally it is based on Chromium and works just fine). In particular, the WebView might be using an older browser engine that doesn't support some modern features the WLED UI uses. Unfortunately, I don't think there is much you can do to use the app on your device - the only thing that comes to mind is that some phones allow you to change the web view implementation (probably in Developer settings). For that, you'd need to tap the build number in the Software information section of your Settings app 7 times and see if you find something like that in Developer settings.

sT0n3r, 2020-01-29:
Thanks for your help, it seems i did not even have webview installed on my phone, after installing from the play store it says i have version 44 but wled app still says Loading WLED UI. here are the test results for my version of webview if it can help in anyway [](https://html5te.st/4b8ec64a07de2a1a)

closed 2020-06-04 by stale[bot]