wled/WLED #635: DMX support

pille, 2020-01-27:
it would be nice to see DMX support (over Art-Net or E1.13).
that way the WLED becomes a fixture and its LEDs could be remotely controlled via standardized controllers and integrated into complex scenes.
it's possible to implement different modes, with a different complexity/amount of channels
the most basic one may treat all LEDs the same, exporting four channels (brightness, R, G, B).
a more complex one would export all LEDs individually (master brightness, R, G, B, R, G, B, ...).
another channel could just trigger effects locally.

sansillusion, 2020-01-27:
E1.31 is already supported just have to activate it in sync settings.

sansillusion, 2020-01-27:
The current e1.31 support gives each led 3 channel (R.G.B.).

Aircoookie, 2020-01-28:
Thank you for the suggestion!
E1.31 is already supported on a per-pixel RGB basis as @sansillusion said!
Art-Net support has already been requested: https://github.com/Aircoookie/WLED/issues/417

closed 2020-01-28 by Aircoookie

pille, 2020-01-29:
thanks @sansillusion for the hint. i successfully played with the current implementation and am planning to extend support for different modes.

pille, 2020-01-30:
that would require adding a new config parameter, which seems rather complex as parameters seem to be relevant in many places. is there some kind of best practices? e.g. this would effect EEPROM layout.

pille, 2020-01-31:
@Aircoookie, can you please help me out with the info requested above?

pille, 2020-02-12:
announced DMX-modes were implemented in #665.