wled/WLED #636: Funktioniert der Wemos D1 Wifi auch?

Cosmicbase, 2020-01-27:
Hallo zusammen,
habe für meinen vorhandenen Arduino Uno bzw. dessen Gehäuse einen neuen Bruder gefunden. Er schimpft sich Wemos D1 Wifi und hat die Maße und Stecksockel vom Arduino.
Er kommt mit dem Chipsatz ESP8266. 

Hier ein Link für alle die es interessiert. 
https://www.makershop.de/plattformen/arduino/wemos-d1-mini-uno/

Aircoookie, 2020-01-27:
Hi, natürlich funktioniert der, WLED läauft auf jedem ESP8266 :)

Cosmicbase, 2020-01-27:
> Hi, natürlich funktioniert der, WLED läauft auf jedem ESP8266 :)

OK Danke :)

Cosmicbase, 2020-01-27:
Achja fast vergessen, welche Binary brauche ich für das o.g. Modell?

Aircoookie, 2020-01-27:
Die ganz normale WLED_xxx_esp8266.bin! Hier ist eine Test-Version die neuer ist als 0.9.0 :) https://drive.google.com/open?id=1GeWCOK1BfCeUIzwl-zL3zzf-gHnhg4zp

Cosmicbase, 2020-01-27:
Super Support hier 5*. Danke

Kann ich die Binary nur per USB einspielen oder auch OTA?
Oder muss ich das 1. Mal nur über USB einspielen? 
Binary = Firmware?

Aircoookie, 2020-01-27:
Danke :) Schaffe ich aber auc nur wenn ich die Frage direkt beantworten kann
OTA geht nur, wenn vorher schon eine Software auf dem ESP installiert ist die es unterstützt.
Bei einem brandneuen geht es deswegen nur über USB.
Ja genau, die Binary ist die WLED-Firmware, man erhält sie durch das Kompilieren des Quellcodes.

Cosmicbase, 2020-01-28:
Was ich noch nicht verstanden habe ist die Sache mit der Libary.
Brauche ich das hier?

Installing with Boards Manager
Starting with 1.6.4, Arduino allows installation of third-party platform packages using Boards Manager. We have packages available for Windows, Mac OS, and Linux (32 and 64 bit).

Install the current upstream Arduino IDE at the 1.8.7 level or later. The current version is on the Arduino website.
Start Arduino and open the Preferences window.
Enter https://arduino.esp8266.com/stable/package_esp8266com_index.json into the Additional Board Manager URLs field. You can add multiple URLs, separating them with commas.
Open Boards Manager from Tools > Board menu and install esp8266 platform (and don't forget to select your ESP8266 board from Tools > Board menu after installation).
Latest release Latest release
Boards manager link: https://arduino.esp8266.com/stable/package_esp8266com_index.json

Aircoookie, 2020-01-28:
Genau, der Arduino Core ist sowas wie die Bibliothek für den ESP8266.
Brauchst du aber nicht, wenn du WLED nur benutzen möchtest und nicht selbst modifizieren/kompilieren. Dafür reicht es, eine Flasher-Software runterzuladen, z.B. esphome-flasher. Ich empfehle das esptool, da brauchst du einfach nur diesen Command zum flashen der Software:

```
esptool.py write_flash 0x0 .\WLED_dateiname.bin
```

Cosmicbase, 2020-01-28:
So etwas wie einen FastLED Sketch brauche ich nicht wie bei Arduino?

Aircoookie, 2020-01-28:
Nein, diese ganzen Bibliotheken brauchst du nur, wenn du WLED als Arduino Sketch kompilieren möchtest (wie man das bei einem normalen Arduino tun würde)

Die Binary nimmt dir diese Arbeit ab, du brauchst weder Arduino IDE noch weitere Libraries abgesehen von einem Flasher-Programm wie esptool :)

Cosmicbase, 2020-01-28:
Danke Aircoookie und schön das es einem Laien leichter als bei anderen Projekten gemacht wird.
Zu guter Letzt noch die Frage wie ist wenn ich die Sache zunächst lokal betreibe über USB und nicht über WLAN.
Wäre dann das o.g. Szenario immer noch gleich?
Oder gild dann eine andere Vorgehensweise?

Vielen Dank :-)

Aircoookie, 2020-01-29:
Hi,
also das Ambilight funktioniert auch gut über USB (Adalight Protokoll). Wenn Hyperion.ng das kann spricht nichts dagegen es zu nutzen. Die eingebauten Effekte und Konfiguration geht immer über WLAN, USB dient nur zum Flashen und für die Stromversorgung :)

Cosmicbase, 2020-01-29:
Ja was ich aber meine das mein Arduino ja nur als Adalight geht wenn ich den FastLED Sketch einspiele.
Wenn ich aber den Wemos D1 Wifi nehme (maßlich gleich Arduino Uno) und per USB als Adalight nutze, dann brauche ich doch einen Sketch oder verstehe ich da was falsch?

Aircoookie, 2020-01-29:
Nein, WLED ist ebendieser Sketch! Nur weitaus komplexer natürlich als der für den Uno.
Binary ist einfach der von der Arduino IDE erzeugte Machinencode der im Endeffekt auf den Wemos bzw. Arduino landet (du könntest z.B. auch den Speicher eines Arduino Uno auslesen und auf einen anderen kopieren, ohne den Sketch zu haben)

Cosmicbase, 2020-01-29:
OK bedeutet Z.B. wemos D1 Wifi kaufen, .bin flas

Aircoookie, 2020-01-29:
Genau, so einfach ist es :)
Die LEDs müssen standardmäßig immer an Pin D4. Um das zu ändern, reicht die Binary nicht aus, da muss tatsächlich in der Datei nbpWrapper.h der LEDPIN verändert werden. D4 ist aber auch der beste Pin, da der ESP8266 hier das LED signal über einen Hardware seriellen Port schickt, was ihn etwas flotter macht als andere Pins.

Cosmicbase, 2020-01-29:
OK aber D4 ist ja nicht der einzige. Ich brauche ja Data, Clock und GND. (APA102).
Gibt es für dich einen klaren Favoriten oder ESP8266 oder ESP8266EX?

Aircoookie, 2020-01-30:
Gut, dann brauchst du die spezielle APA102 Binary `WLED_0.9.0-b1_ESP8266_apa102.bin
 `von der [Releases-Seite](https://github.com/Aircoookie/WLED/releases).
In dem Fall ist Data D4, Clock D3 und GND ist immer auch GND am D1 mini.

Cosmicbase, 2020-01-30:
Genau das habe ich ja gemeint. Wie WLED weiß was für einen LED Typ ich habe und die Pin Zuordnung. Das müsste man bei Arduino im Sketch einstellen.

Favoriten ESP8266 (80Mhz) oder ESP8266EX (160Mhz
bzw. persönliche Erfahrungen!

Aircoookie, 2020-01-30:
Jeder ESP kann 160MHz, das lässt sich softwareseitig einstellen. WLED läuft immer mit 80MHz, das spart Strom. Wemos D1 mini hat soweit ich weiß den ESP8266EX verbaut. Ist aber egal, wichtiger ist der board-Formfaktor. Ich nehme persönlich nur die D1 mini, weil der Nodemcu wegen der angelöten pins mehr platz braucht

Cosmicbase, 2020-01-30:
OK verstehe. Du lötest dann am Mini nur die Pins an wo du brauchst.
Gar nicht so verkehrt. :+1:

Der Wemos D1 Wifi ist zwar groß. Aber in meinem Fall auch nicht schlecht, weil ich meinen Arduino Uno im transparenten Acrylglas durch den Wemos austauschen kann. (Thema Staub und Schutz).

Aircoookie, 2020-01-30:
Genau :)

Der D1 mini ist doch das kleinste Board mit USB? Du kannst natürlich auch den "rohen" ESP8266mod oder 8266ex nehmen, aber viel Spaß an die winzigen Pins Käbelchen fürs Flashen anzulöten. Dann gibt es noch viel zu beachten mit Reset und dass einige Pins beim Starten High oder Low Signal haben müssen. Insofern würde ich immer nur Boards mit integrierter USB-Buchse empfehlen, da hat man diese Probleme nicht.

Cosmicbase, 2020-01-30:
Jup der D1 Mini ist der kleinste mit USB Buchse. Die 3 Pins anlöten ist wohl das kleinste Problem. Habe gerade mal überlegt ob man den Mini in eine Streichholzschachtel packen kann?

Aircoookie, 2020-01-30:
Sollte passen.

Cosmicbase, 2020-01-30:
Das ein ESP8266 Leistungsstärker ist st als ein Arduino ist ja klar. Aber merkt man bei schnell belebten Bildern ein Unterschied? Baudrate, Hz, FPS.

pille, 2020-02-03:
@Cosmicbase: passen wuerde es, aber wenn du die LEDs ueber den wemos mit strom versorgst, wird der spannungswandler recht warm.

Scheisbolle, 2020-02-08:
> 
> 
> Jup der D1 Mini ist der kleinste mit USB Buchse. Die 3 Pins anlöten ist wohl das kleinste Problem. Habe gerade mal überlegt ob man den Mini in eine Streichholzschachtel packen kann?

Ich mache die Teile in Tic Tac dosen, die gibts ja auch in klein und groß.

closed 2020-03-09 by Aircoookie