wled/WLED #638: xy-color mode?

fribse, 2020-01-28:
Hi Aircookie

I'm not sure about if WLED supports this? Or if the limitation is in the integration to HA?
I have set up a WLED over my sons desk. I've added an IKEA controller (the round 5 button) to control it through deconz and [controllerx](https://github.com/xaviml/controllerx).
It works brilliantly for on/off/dim/brighten, but not the color cycle which is xy-color.
I hope it can be implemented, as that would be a great way to have simple controls of the day-to-day operation of the light.

Aircoookie, 2020-01-28:
Sounds good and shouldn't be that hard to implement, because I already have a XY to RGB converter for the Hue sync. Does the software you use interface to the JSON API?

fribse, 2020-01-28:
I beleive it does, but I will ask @frenck on HomeAssistant github :-)

frenck, 2020-01-28:
From Home Assistant, it doesn't make much sense IMHO. Home Assistant already accepts hs and xy colors for WLED.

fribse, 2020-01-28:
I see, that is interesting, because when I do the traversing with controllerx I see an error about that in the log. I'll retest with setting it specifically to xy colors in the config.

fribse, 2020-01-29:
I've tested it, and if I set color mode for the light, it WILL traverse the colors as it should.
So now my son can control his desktop lights completely with his IKEA controller for normal use and for effects he uses the WLED app :-)
It also works via Google, though for some reason he can't say 'set desktop to warm white' but has to say 'change desktop to warm white' :-)

closed 2020-01-29 by fribse