wled/WLED #639: Filesystem

jure3000, 2020-01-28:
Why does WLED not use filesystem by default (for serving html, saving settings)? What is the advantage of EEPROM libary, if it uses flash just like filesystem?

Def3nder, 2020-01-28:
Hi @jure3000, one advantage probably is that is "sonsumes" only 2560 bytes of the flash.
For a filesystem you would need a partition that is at least 64k big (see [here](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html)):
```
|--------------|-------|---------------|--|--|--|--|--|
^              ^       ^               ^     ^
Sketch    OTA update   File system   EEPROM  WiFi config (SDK)
```

On a chip with 4MB flash this is not a problem, but on a 1MB chip, this would lead to OTA not being possible anymore.

Aircoookie, 2020-01-28:
Yep, what @Def3nder said is correct! No OTA on 1M modules once we use SPIFFS.
Starting in 0.10.0, I will probably be using it though, because the current preset system is very limited by EEPROM (only 16 configs, and only a single one can save segments) and a filesystem would allow for easier backup and restore of the configuration. For hosting custom html, you can already enable it in wled00.ino by setting the defines from line 28 accordingly:
```
//#define WLED_DISABLE_FILESYSTEM    //SPIFFS is not used by any WLED feature yet
#define WLED_ENABLE_FS_SERVING   //Enable sending html file from SPIFFS before serving progmem version
#define WLED_ENABLE_FS_EDITOR    //enable /edit page for editing SPIFFS content. Will also be disabled with OTA lock
```

jure3000, 2020-01-28:
Thank you for clarifying. Are there many 1MB modules in use? Since 4MB are so affordable, I don't know why would someone go for 1MB.
Does that mean that in 0.10.0, OTA support for 1MB modules will be removed?
When do you plan to release 0.10.0?

Def3nder, 2020-01-28:
Hi @jure3000,
there are quite a few ESP8285 around and these "special 8266 modules" have an 1M chip.
And: if you wanna build a tiny little thing, you probably want to use a ESP01, becuase it's only 1x2cm big :-) - if you don't want to desolder the SMD flash chip and solder a 32Mbit one instead, you are stuck with either 1M or even 512k 😄

closed 2020-06-03 by stale[bot]