wled/WLED #644: Wifi/connection problems

toyotaspeed90, 2020-01-30:
So I'm new but not completely ignorant.

I have 13 Nodemcu chips, three branded Amica and 10, well not.

Around the house as a test they worked fine with a random error here or there using WLED on my Galaxy S7.

Then I prepped for larger scale.  Installed a box with two 300W PSUs, a beadboard with 3 nodemcu chips (all with WLED), and a sonoff basic on the side of the house (close proximity to the router, 20ft, same side/wall).

Constant problems connecting.

WLED will state the 3 chips are online but then cant load the UI stating the device is unreachable.  It will then show the devices as offline.

Fing will see the devices as connected- though isn't uncommon to see one drop.

The sonoff in the same box is always available.

When it works I can be 200-300ft from.the house (the opposite direction) and everything works.  However this is rare and at times I spend 20 min trying to connect by restarting the systems/psus.  The sonoff always works - again, in the same enclosure.

I also have an rpi with Hassio (that's been spotty lately), a shelly 2.5 without issues, and an RM3 that's good.

Lastly-the router is a Cisco/netgear dual band with 5ghz turned off (and even renamed to ensure nothing can connect). Pinging the nodemcu chips gets mixed results but the sonoff is stellar.

I used the most recent release (December) .bin file.  Have tried both brand of hardware with identical problems.

I hope someone is able to assist to resolve this.

Def3nder, 2020-01-30:
Hi @toyotaspeed90, sounds wired - you could check the channel the ESP's are connected to 
(http://[WLED-IP]/json/info 
you get a JSON reply among others with an object 
```
"wifi": {
    "bssid": "7C:FF:4D:1D:EB:B1",
    "rssi": -52,
    "signal": 96,
    "channel": 11
  }
```

Other people reported that **changing the channel to 11** on the router did result in a better connection.

Another thing I would try is to flash all three WLEDs with Tasmota (I asume that's what is on the SonOff, too) and look if then all four are perfectly connected.

Does this make sense ? From there we could further try to solve this issue 😄

toyotaspeed90, 2020-01-30:
Makes sense.  I had a few minutes this morning before heading to work and I plugged one of the offenders in close to the router - they're on channel 3.  I didnt immediately see a way to change the channel on my router GUI but. Like Isaid, only had a few minutes.

I should add - the IPs are all static.

Another question - are these all handled locally or IOT? While our network/router seems to have great connectivity, our internet is less than exceptional.

huggy-d1, 2020-01-30:
What type of enclosure (material / metal/plastic) is housing the node mcu's?  If it is metal, you need a different module, one with an external antenna jack so you can install antenna connections to the exterior of the metal enclosure - and add 1 antenna for each module.

toyotaspeed90, 2020-01-30:
Its plastic.  There are 2 300w PSUs at the top, the sonoff between them. Then a breadboard at the bottom with the nodemcu's.  I could probably take a 20ft cord, if I were to drill a hole in the wall, and get from my router to the nodemcu.

The reason I keep mentioning the sonoff is because they utilize an esp8266 chip.

toyotaspeed90, 2020-01-30:
[image]
 Here's before going on the wall and before the sonoff basic.

toyotaspeed90, 2020-01-31:
Got home - swapped to channel 11.  I was able to connect - them went to LED Preferences and it hung up - device no longer reachable.  Repeatedly works then doesn't.  Channel 11 seems to help but only marginally.

So I remembered I have another Nodemnu with Robs Home Automation led program on it (Which I could never get quite right with Hassio - and why I went to WLED and the computer doesnt recognize the device to reflash).

Plugged it in near the router to get the IP, did some ping tests - all great.

Went outside - plugged it in.

Ping tests run awesome.

Hmm...

Run ping tests on the WLED nodemcus... they all run awesome.

I go into WLED and it works for about 30 seconds - fails continuously in the segments page, then the device becomes unreachable.

toyotaspeed90, 2020-01-31:
This is all within 2 minutes of each other (should be noted - WLED says its offline and I ran the ping test in Fing right after).
[image]
[image]
[image]
[image]
[image]

toyotaspeed90, 2020-01-31:
Try again on the last 2 screenshots...

[image]
[image]

To note.... roofline1 is unplugged.  Wledtest is its replacement, roofline2 has the lights, roofline3 is also plugged in, and the light test is the non WLED nodemcu

huggy-d1, 2020-01-31:
I wonder of the beta version with segments is the problem. How many segments?  Have you tried the last official release without segments to see if that’s stable?

snook, 2020-01-31:
I was just coming to post about issues with losing WiFi as well, not positive it's the same issue, struggled with adding to this ticket or creating a new.

I am using WLED 0.9.0-b1

My main unit is WLED-G, it is connected to my light strip and it was losing connection about once a day. I have since setup a second NodeMCU (WLED-B) it has no lights running, it is just on may garage workbench powered by a USB charger as a test to see if my main NodeMCU issues were from interference from being near the power supply or an issue with the unit itself. Today they both lost connection at exact same time as seen in screenshot for Home Assistant. I have also added a mesh router in the garage to ensure they are getting good WiFi signal. Using Asus GT-AX11000 and RT-AC68U as a mesh extender.

The router and Fing show they are still connected to the network but they are not responsive when I ping them or try and load the config. I saw the suggestion to change to channel 11 in another ticket and tried that as well with no change in performance.

They also will never reacquire a connection to WiFi if I simply reboot the router, I have to cut the power and reboot them to reacquire a connection. I was going to report this as a ticket, but maybe it is related. They show a lot of short disconnects in Home Assistant throughout the day while all my other devices show pretty solid activity.

[image]

toyotaspeed90, 2020-01-31:
The Json file says its version 0.9.0-b1

Maybe I'll try a previous release and check stability...

Def3nder, 2020-01-31:
Hi @toyotaspeed90, I just saw your 3 ESPs next to the huge PSU's:

in other forums they reported **issues if the ESP is too close to an PSU**.
This problem gets worse with the "quality of the PSU" - so e.g. some cheap AliExpress ones result in more problems than ones from Meanwell. 
You get these kind of problems with switched-mode power supplies - the old legacy linear power supplies with a transformer and a rectifier did not produce this high frequencies. The switching power supplies do produce a lot of **interfering high frequencies in the GHz range** that result in WiFi disconnection.

So I would suggest two things:

1. Add a `1000uF capacitor` in parallel to the power lines next to the ESPs (one for each PSU)
2. (if possible) take the `ESPs away from the PSUs`. 

**As a test** would it be possible to grab that SonOff you mentioned an put it in the place where the three ESPs are located now ? ...and see if the SonOff get's the same issues when residing so close to this big PSU's ? 
I think you would need to add the SonOff there (instead of replacing the ESPs), because we need some load on the PSU (coming from the WLED-strips).

toyotaspeed90, 2020-01-31:
The sonoff resides on the plywood directly between the 2 PSUs, so itstecnically closer.  Only 1 PSU is actually being used at this time.  Theres currently only 1 roofline with 110 LEDs installed - so while the PSU has a high capacity only a small portion is being used.  I had done a test with a shelly 2.5 in the house with a prop having 225 LEDs and the power consumption was dependent on the brightness/settings of the LEDs, meaning the PSU doesn't put out its capacity, it puts out (and pulls from 110) what is being drawn from the attached components.

I find this an interesting response since the 4th Nodemce without WLED was thrown on there for a test and isn't having the same problems, nor is the sonoff.

I'll think about it.

toyotaspeed90, 2020-01-31:
On my drive to work I thought about this... wouldn't the capacitor fix (if an issue) the "quality" or frequency of the power supplied to the Nodemcus but would have no effect on the possible proximity issue...

Def3nder, 2020-01-31:
Hi @toyotaspeed90, no the **capacitor would only fix power fluctuations**. They would lead to a reboot in the worst case. The "bad quality" of the PSUs if more **a frequency problem that directly interferes with the Wifi Signal**. This could only be "fixed" in the PSU design (and that does cost more money) - this is why manufacturer like Meanwell need to use higher prices...

biscuitgod, 2020-01-31:
I JUST started playing with WLED and had this same connectivity issue with 0.9.0_b1.   I had a spare MCU laying around so flashed it with v0.8.6 and have had no issues since.   I think there's definitely an issue in the newest firmware.

toyotaspeed90, 2020-02-01:
Interesting.  I ordered a FPP F16 just because come October I cant have these issues on 1, let alone 10+.

Anyways- as the tinkerer I am (and I really do like the new/beta version) I reflashed one to the previous version and something odd happened.

The reflashed previous version works 99% flawlessly (the one time it didn't I think Hassio was trying to ping it in the background while I also opened WLED).  But no issues.

The oddity is the other 2, with the new beta 9 version - no longer have problems.  It's also much faster.

I literally changed nothing else.

So - it makes me wonder if there's an issue with the wifi on the current release that is increased with additional nodemcus.

One major suggestion I would make with Segmets (since when they work it's so much better) - is make them part of the LED preferences configuration.  My setup, and I can't imagine I'm alone, has the Nodemcu turning on and off, and when it recycles ( at least for me) all segment information was lost.

One Nodemcu and 1, maybe 2 segments, isn't a big deal.  If the plan, however like mine was, is to ha e muktiple nodemcu's with multiple segmets - updating that each time would get old.

Def3nder, 2020-02-01:
Hi @toyotaspeed90, the segments can (right now) only be saved to preset 16.
So the best way having a configuration with multiple segments is to save this in slot 16 and assign "16" to the boot marco.

toyotaspeed90, 2020-02-03:
3 days of zero wifi connection issues (no other changes) to the Nodemcu with the earlier release.

RAMBORG, 2020-02-06:
WiFi will now auto-reconnect if connection lost
I think this option is not included on the newest version.

spiff72, 2020-02-08:
I just wanted to check in here.  I just discovered WLED today, and spent 30 minutes trying to figure out why my Node MCU was not staying connected to my network for more than 5 seconds.

I was using the latest beta when I found this issue report.  I now tried 0.8.6 and it worked perfectly on the first try.

toyotaspeed90, 2020-02-09:
Still experiencing issues with the 9+ release and none with the earlier release.  They are housed in the same enclosure still, so I have a hard time thinking it is hardware related.

Def3nder, 2020-02-20:
Hi @toyotaspeed90, @snook, @spiff72, 
could you try to disable UDP notification and E1.31, DMX etc. - basicaly everything that does use UDP ?

Are the disconnects still there ?

@Aircoookie: what about sending a PING to the default gateway every 10 seconds ?

...with this code we could implement an own PING function:
```
void ICACHE_FLASH_ATTR user_ping_recv(void *arg, void *pdata) {
    struct ping_resp *ping_resp = pdata;
    struct ping_option *ping_opt = arg;

    if (ping_resp->ping_err == -1)
        os_printf("ping host fail \r\n");
    else
        os_printf("ping recv: byte = %d, time = %d ms \r\n",ping_resp->bytes,ping_resp->resp_time);
}

void ICACHE_FLASH_ATTR user_ping_sent(void *arg, void *pdata) {
    os_printf("user ping finish \r\n");
}

void ICACHE_FLASH_ATTR user_test_ping(void) {
    struct ping_option *ping_opt = NULL;
    const char* ping_ip = "192.168.1.114";

    ping_opt = (struct ping_option *)os_zalloc(sizeof(struct ping_option));

    ping_opt->count = 10;    //  try to ping how many times
    ping_opt->coarse_time = 2;  // ping interval
    ping_opt->ip = ipaddr_addr(ping_ip);

    ping_regist_recv(ping_opt,user_ping_recv);
    ping_regist_sent(ping_opt,user_ping_sent);

    ping_start(ping_opt);
}
```
 _source: https://bbs.espressif.com/viewtopic.php?t=583#_

spiff72, 2020-02-20:
I am not sure how to perform the suggested adjustments above - are these in the WLED settings pages?

Also, I may have been incorrect in my original post - I initially installed the **WLED_0.9.0-b1_ESP8266_1M_full.bin** binary when I was unable to get a connection to my router.  I tried again a few days later with the **WLED_0.9.0-b1_ESP8266.bin** binary, and it seems stable.  I haven't noticed the failure to reconnect after losing wifi, but I haven't played with this version much, as I see some other odd issues in beta (some effects seem to run more slowly than they did on the 0.8.6 version, and I see a lot of error message popups in the interface).

Def3nder, 2020-02-20:
@spiff72, yes, it's in the settings:
Settings -> Sync Interfaces:
uncheck everything here:
[image]

...and everything here:
[image]

Yes, the error messages appear if you change things in the UI "too fast": every request needs some of the heap (RAM) and it takes a moment to free the used space.
If the ESP cannot allocate the needed memory, it cannot answer the request comming from the UI and the UI does display an error.

spiff72, 2020-02-20:
Thanks - I will adjust these later tonight.  Has the apparently slower performance of some of the effects been mentioned at all here (or is this my imagination)?  The one that seems the most affected to me is the Fire2012 effect.  It just doesn't seem as fast as the previous version, even with adjustments being made to the speed and intensity settings.  Others like the sweeps and wipes don't seem as fast either.

xavieraijon, 2020-03-10:
I have many disconnection problems. I have tested with DHCP, with fixed IP, activating the new WIFI sleep function of version 0.9.1 and nothing. I have to do several reset, it has been online again, but it lasts minutes, sometimes hours, sometimes 1 whole day, but always disconnects again ... I don't understand why.

raoulteeuwen, 2020-03-10:
@xavieraijon Can you rule out 'normal' wifi problems? What's the distance between your AP and the nodeMCU/Wemos/... ? Any obstacles? Other wifi networks in the vicinity? Other wifi obstructing tech? Can you bring the controller closer to the AP to see whether that helps?

xavieraijon, 2020-03-11:
I have no other problems with wifi, the distance is quite short, they are close. I will try to put it even closer, but I think that is not the problem ... Is there any log to see when and why it disconnects?

Thank you!

jdubz666, 2020-06-26:
Yup I'm at the point where I'm refusing to get up and unplug and replug my ESP's. 

Seems like Wifi dropping will always happen and you have to live with it.

My solution... I'm using Smart Plugs, when home assistant detects Wled is unavilable, it turns the plug off and back on.

If anyone has anything more elegant, I'd love to hear.

jdubz666, 2020-06-26:
> I have no other problems with wifi, the distance is quite short, they are close. I will try to put it even closer, but I think that is not the problem ... Is there any log to see when and why it disconnects?
> 
> Thank you!
> 
> EDIT: Maybe it's a coincidence, but since yesterday I did the last reset, I was still online. I have integrated it into Home Assistant, and losing the connection ... can it be the cause? Bad integration with HA?

I also have the Home Assistant integration and have many Wifi drops. Did you try testing it without Home Assistant integration?

closed 2020-10-31 by stale[bot]