wled/WLED #645: Alexa/Echo listening effect

gahujipo, 2020-01-31:
We'd like to use an Alexa listening effect in a small theatre project where we'd like to imitate the blue ring and the green "dot" while an Amazon Echo is listening. I thought about using an ESP8266 and a small piece of WS2812B which i have here left from another project. 

From the documentation I wasn't able to understand whether there is a built-in macro or effect which looks like this. 

At the end I'd like to trigger this somehow over the network from our controlling software which is capable of running something on the command line (e.g. a curl or a netcat). 

Is there a built-in effect which looks like the the light which is shown on an Echo while it is listening to voice commands?

huggy-d1, 2020-01-31:
https://github.com/Aircoookie/WLED/wiki/List-of-effects-and-palettes might help

gahujipo, 2020-01-31:
Thank you. Yes, I was already carefully reading that page. Could you spot such an effect?

huggy-d1, 2020-01-31:
I did not see one right off, but I guess it depends on the way the strip is wired as to how to would look and feel relative to the desired effect.  Say you wire it starting at 12 o'clock looking down on the "dot" compared to 6 o'clock.  Different effects are needed for each clock position to achieve the desired results.
Or, wire up 2 strips at the 6 O'clock position, one going clockwise and the other counter.
Then you only need to solve the problem of wiping from end to start and have the start behavior look like the dot when it is listening.

I hope that made sense.

Just try to solve 1/2 the problem and wire 2 short strips starting at the 12 or 6 O'clock position and find one effect that handles the left side and the right side will follow suit (if I recall how the dot lights up starting at 12, going both CW and CCW away from 12 towards 6 - after initially lighting up a certain way.

huggy-d1, 2020-01-31:
Either that, or use 2 WLED controllers to handle the essentially two different dot lighting effects

gahujipo, 2020-02-05:
I am planning to write an own effect which could do this in some way, but I would need a little help to understand how to start. 

My plan for now is to create an effect which builds up a half blue circle from LED1 to Count(LEDs)/2 and at the same time  from Count(LEDs)/2 to Count(LEDs). Then set the color of the first 5% to green. 

My question is how to I configure my method in FX.cpp to stop after the two half circles are fully lit up?

huggy-d1, 2020-02-07:
You can create your own effect(s) in your usermod.  There is at least one way to take control of the LEDs temporarily (or permanently) within a usermod by temporarily disabling the automated LED updates.

Aircookie posted about it in one of the issues somewhat recently (think it is still open).

If you can not find it, post back.

closed 2020-06-13 by stale[bot]