wled/WLED #646: dropping connection after 1 hour

Dareno1995, 2020-02-01:
Hello guys,

So i searched the forums and google for a solution but cant find anything about this problem.
I use a Wemos D1 R2 with the W-led pre compiled bin.
This is connected to a W2812B led strip.
I got this connected to Home assistant trough the Wled integration.
This works wonderful with nice colors, except one thing.

It turns off out of nowhere after 1 hour then its not available through Home assistant for 1 hour.
After that i comes back online.
It is reachable in the mean time through the IP adres where i can switch it on and it becomes available again, also in Home assistant.

Also when Im not connected to Home assistant it keeps switching off after 1 hour.
See the attached picture for the status through home assistant.
Here you can see it's +- 1 H switching.

I think it's just some configuring error, but couldn't find out what.
Suggestions would be awesome!

[image]

RAMBORG, 2020-02-06:
Lost WiFi connection?

Landrash, 2020-03-24:
Try wiping your esp and reinstall.
Had a similar problem with one of mine that I couldn't figure out.
Wipe over usb and then reflashed with wled and I've never had a problem since.

closed 2020-07-29 by stale[bot]