wled/WLED #647: Alexa app

vinit-bhati, 2020-02-01:
i have an issue here i don't have alexa device but i have the alexa app installed and i am not able to discover my wled lights i have setup of ws2811 50 leds please tell me if i can run this in the alexa app so please help me

Aircoookie, 2020-02-01:
Hi,
I am very sorry, but the Alexa emulator only works in the local network over the Hue protocol. You will need a physical Echo device to make it work :( Just the app will not be able to discover or control WLED devices, but once you have a physical Echo control via the app would also be possible, as it is relayed through the Echo device.

vinit-bhati, 2020-02-01:
oh man but thanks you helped me a lot and i really loved your work please
more things in it it will be freaking awesome and one more thing i am not
one option in the clock menu one is analog clock the middle one is not
showing up

vinit-bhati, 2020-02-01:
see

Aircoookie, 2020-02-01:
Thank you :) Yeah, the option below analog clock is grayed out because I removed it, but needed to keep the entry for compatibility reasons

closed 2020-06-07 by stale[bot]