wled/WLED #648: 4-Relays connected

sjude68, 2020-02-01:
@Aircoookie I have used 4-relays to turn on/off 4 devices with your previous help.  Added Alexa and MQTT modifications. 

When I toggle LED strip the Relay2 (device3) also gets toggled through Alexa or the GUI. 

I have attached the wled00 with my modifications. 

[wled00.zip](https://github.com/Aircoookie/WLED/files/4142725/wled00.zip)

Aircoookie, 2020-02-02:
Hi, this is the relay feature that is interfering :)
WLED offers support for a relay pin that turns on and off with the strip. This can be used to save standby power by cutting the LED supply. This is set to GPIO12 (D6) by default, the pin you are using for Relay2.
Set `#define RLYPIN` to -1 in NbpWrapper.h, that should fix the issue!

sjude68, 2020-02-05:
Thanks @Aircoookie. How do I move the modifications to usermod so when there is an update I can just recopy the usermod file to get the new version active.  I mean the modifications in Alexa and MQQT files in one location.  Is it possible.

WLED is great and it is a beautiful share. Thanks @Aircoookie.

Aircoookie, 2020-02-06:
@sjude68 this is a good point. It should be possible to copy at least the custom Alexa devices. MQTT is a little more difficult though. I think it would probably be best if I added a few functions to the usermod file, maybe these
```
userInitEspalexa
userPublishMqtt
userSubscribeMqtt
userJsonReceive
userJsonSend
```
This would make it possible for the usermod to interface to Alexa, MQTT and JSON without modifying other WLED files, thus allowing an easier WLED update!

sjude68, 2020-02-07:
I moved the alexa lines and also the mqtt subscribe lines into the usermod
sections for UserAlexa as suggested.

Not sure how to move mqttpublish lines. I am not using Publish.

sjude68, 2020-02-13:
@Aircoookie if I use the same wled code which controls led strip and 4-relays to another nodemcu board it gets duplicated.  I changed the AlexaInvocation name in the code to mark it for different device names but relay1 on 1st board turns relay1 on the 2nd board also.  The led strip work differently.

deppen12, 2020-03-22:
https://www.hackster.io/pawan-kumar3/google-assistant-based-relay-controlling-c45f85
can we use this with wled, coz it has google assist . i dont have alexa so i want to use google . and also make web ui for relay like wled

austwhite, 2020-03-22:
You can hook WLED to another he assistant hub via MQTT and use Google.
I do that with Home Assistant currently.
There are much better projects for Relay control though.
WLED is designed for LEDs.

closed 2020-05-26 by sjude68