wled/WLED #649: Alexa Color Adjustment

TechnoHoschi, 2020-02-01:
Hello,

how can I adjust the colors or gamma for Alexa? Especially the white tones contain too much red. In version 0.8.6 I could adjust this in "WS2812FX_fcn.cpp". In the current version it should probably be "EspalexaDevice.cpp". Which part of the code should I change for gamma adjustment or color calibration?

Thanks and Regards
Michael

Def3nder, 2020-02-03:
Hi @TechnoHoschi, yes, the colors are calculated in EspalexaDevice.cpp.

Alexa pushes them in CT (color temperature) to WLED and there they are converted to RGBW.

If you like to change the RGBW values, lines 125++ is the right place to do:

```
// Cold white to warm white receiving from Alexa: _ct = 199, 234, 284, 350, 383 (from cold white to warm white)
    switch (_ct) {
      case 199: rgb[0]=255,rgb[1]=255,rgb[2]=255;rgb[3]=255;break;
      case 234: rgb[0]=127,rgb[1]=127,rgb[2]=127;rgb[3]=255;break;
      case 284: rgb[0]=0,rgb[1]=0,rgb[2]=0;rgb[3]=255;break;
      case 350: rgb[0]=130,rgb[1]=90,rgb[2]=0;rgb[3]=255;break;
      case 383: rgb[0]=255,rgb[1]=153,rgb[2]=0;rgb[3]=255;break;
```
...while writing this I just see that the values will only fit for RGBW and not for RGB.
I will put it on my "[to-do list](https://github.com/Def3nder/WLED/projects/1?fullscreen=true)" to look into this.

TechnoHoschi, 2020-02-04:
Thank you very much,

Take your time, I've already ordered RGBW Leds. ;)
But because of the current situation in China, it may take a little longer anyway.

greetings

Michael

TechnoHoschi, 2020-03-28:
switched to sk6812ww and all colors are fine

closed 2020-03-28 by TechnoHoschi