wled/WLED #650: Allow for longer transition times

chaptergy, 2020-02-01:
Hey,

my specific problem is, I'm trying to create a light alarm using WLED leds using my home assistant. However in the wiki I read, that the max transition time is 65000ms, so about one minute. For this light alarm I'd want to set the transition to maybe 20 minutes. What was the reason for capping the transition time at this value?

Thanks!

Aircoookie, 2020-02-02:
Hi,

yes, this is true because the transition time is stored in a 16-bit integer, so no transition times longer than about 65 secs are possible. I could change this in the future to reduce the precision to 100ms/step (which the JSON API currently uses), which would yield a transition range of up to 6500 seconds, so almost two hours.

That said, for now you might want to look into the Nightlight function :) You can set a duration and target brightness and it will essentially create a very slow brightness transition for you (1-255 minute range)

Def3nder, 2020-02-03:
Hi @chaptergy, please try the "sun rise" feature of the NightLight for this.
Sun Rise is like Nightlight but with additional color-transition from the primary to the secondary color.
paste this string to one of the macro slots:

`&FX=0&CL=hFF7800&C2=hFFC28C&W2=255&A=5&NT=255&NL=20&NF=2`

It does the following:
- `&FX=0` ... set the Effect to "Solid Color"
- `&CL=hFF8C00`  ...  set the primary color to Redish
- `&C2=hFFC28C&W2=255`  ...  set the secondary color to a warm white (change this if you do not have a dedicated white color channel)
- `&A=5`  ...  set the starting brightness to 5 (about 2%)
- `&NT=255`  ...  set the target brightness to 255 (100%)
- `&NL=20`  ...  set the transition time to 20 minutes
`&NF=2`  ...  set Nightlight to "On with color transition"

Is this what you'd like to have ?

chaptergy, 2020-02-03:
Hey,
thank you very much for your suggestions. However it seems this is unfortunately not possible through the home assistant integration.

closed 2020-06-09 by stale[bot]

hansenchristian, 2021-04-18:
@chaptergy  I use home assistant too and I send my request over mqtt, works with @Def3nder 's solution perfectly
`service: mqtt.publish
data:
  topic: wled/test/api
  payload: '&FX=0&CL=hFF7800&C2=hFFC28C&W2=255&A=5&NT=255&NL=2&NF=2'`

chaptergy, 2021-04-19:
> @chaptergy I use home assistant too and I send my request over mqtt, works with @Def3nder 's solution perfectly

Yes, his solution is a great workaround which works for me, too.

daguej, 2026-09-23:
For anyone interested, I've posted a [patched version of WLED](https://github.com/daguej/WLED/releases) that supports transitions up to 24h.

With this patch running on your WLED device, changes triggered by Home Assistant with long transitions just work as expected.

softhack007, 2026-09-23:
Please keep in mind that this patch is very special-purpose. It was not accepted into the WLED core due to side-effects on other WLED features - especially wled sync won't be able to synchronise transitions on several devices any more.