wled/WLED #652: How to turn on/off a specific segment without turn pn/off all others

nmsoares, 2020-02-02:
I'm running version 0.9.0-b1 and I've setup segments ... Very Cool!...but how to turn on/off a specific segment without turning on/off all others ???

brentbrooks70, 2020-02-02:
@nmsoares this is close to #511, although it is in use with API. From the GUI it is probably easiest right now to set it to solid black. If you want it to maintain colors and effects then deleting the segment will turn it off and redefining it will turn it back on to previous config as per #511.

nmsoares, 2020-02-02:
Thanks for the information @brentbrooks70 .Can you share the command ? I'm stuggling to find the MQTT commands to set to black on a segment.

Def3nder, 2020-02-03:
Hi @nmsoares, 
unfortunately you can only change segements (other than the 1st one) through the JSON API.
So you need a client that can send a JSON POST request to WLED.

Aircoookie, 2020-02-03:
There actually is a command in the HTTP to select the segment to which to apply changes: `win&SM=[segment number]`. I have to admit though that it is not well tested and may have side effects, that's the reason it isn't documented right now.

Def3nder, 2020-02-03:
Thx @Aircoookie ! 

@nmsoares, as the segment commands are parsed in the very beginning this should work then:

Assuming the we have 3 segments: `seg0 {0 to 50}`, `seg1 {50 to 100}` and `seg2 {100 to 150}` and we want to turn off seg1:
`http://[WLED-IP]/win&SS=1&S2=0`
or send a payload with `win&SS=1&S2=0` to `/WLED/[WLED-name]/api`

For re-enabling the segment your need to know the segment-end - this command would do the trick:
`http://[WLED-IP]/win&SS=1&S2=100`

Doing this should not change any color or effect running on that segment - this should all stay in place.

nmsoares, 2020-02-03:
Thanks @Aircoookie  & @Def3nder ! TOP!!!!

Def3nder, 2020-02-05:
Hi @nmsoares, could you pls. close the issue, if you are satisfied with the results ?
Thx a lot !

closed 2020-02-07 by Aircoookie